canboat/canboat #454: AIS Binary Messages - in more detail

dwpsco, 2024-08-28:
https://navcen.uscg.gov/ais-addressed-binary-message6
https://navcen.uscg.gov/ais-binary-broadcast-message8

keesverruijt, 2024-08-28:
And your point is?

dwpsco, 2024-08-28:
Thought you could add it to the documentation... But hey no worries.

Do you still want a log file for PGN129538? I also have log files for several propietary Furuno devices if you want them, if I'm not bothering you.

keesverruijt, 2024-08-28:
Yes, a logfile for PGN 129538 would be useful.

For proprietary PGNs we can't really do much with logfiles, we need to know what the messages "do" for it to be useful for others, so why don't you spend some time and try to reverse engineer them? It's the best way to repay for open source ... write some yourself! A compiler suite is easy to get nowadays, even on Windows.

dwpsco, 2024-08-28:
Yes, I have been working on reverse engineering them. I have attached a csv for 129538, Source Address 3 is my Furuno Satelite compass.

[N2K129538.csv](https://github.com/user-attachments/files/16787918/N2K129538.csv)

larsujensen, 2024-09-28:
A comment about the us coast guard documentation. The AIS standard is an ITU standard, current revision ITU-R M.1371-5 I believe. As ITU wants standards to be used they are in my past experience either free or possibly reasonably priced. M.1371-5 is free to download. The problem is nmea, it is difficult to know how they mangle the data.

commit 35fcf417 (2024-12-25): Fix #454: PGN 129538 fields
closed 2024-12-25 by keesverruijt