canboat/canboat #455: BitOffset for repeating fields

inquestruss, 2024-09-09:
For repeating fields the BitOffset is only correct for the first repeat group. Fields with Order >= RepeatingFieldSet1StartField should have BitOffset set to 0.

keesverruijt, 2024-12-25:
Hmmm so you are saying, since the BitOffset value is only correct for the first occurrence of the repeat group, it should not be included at all? 

Even if I agree to that, which I think needs more input from you as to why your point of view is more valid than the current point of view, which is that showing the offset for the first occurrence adds information, the solution would be to remove the BitOffset for this field, not set it to zero.

inquestruss, 2024-12-25:
Missing or zero works for me--golang defaults to zero values for missing json values. I use bitoffset as a validation when reading fields, and it's complex to have it for the first repetition and ignore it for subsequent ones. The current implementation already zeros it for repeating fields--I stumbled on this again implementing some testing code. I'm fine with closing the issue. And Happy Holidays!

closed 2024-12-25 by inquestruss