canboat/canboat #458: How to properly use rate of turn i PGN 129038

larsujensen, 2024-10-05:
Issue #162 brought up that this parameter was not complete in canboat. The issue was not solved at that time. Now, if the rate of turn parameter in PGN 129038 is properly used it is not a major issue. Improperly used it is my belief that it may in rare cases give a wrong rate of turn value. As it is far from obvious how to use it I will try to shed some light on it for a broader audience than last time, and actually also propose how this parameter possibly can be completed in canboat. All that follows is obviously to the best of my knowledge, and it may already be well known, I have not been following canboat for several years.

The value of the parameter comes from AIS. AIS is specified in  ITU-R M.1371-5, and the parameter in AIS is called ROTAIS. The specification of ROTAIS is as follows, directly copied from the standard:

”0 to +126 = turning right at up to 708° per min or higher
0 to –126 = turning left at up to 708° per min or higher
Values between 0 and 708° per min coded by
ROTAIS = 4.733 SQRT(ROTsensor) degrees per min
where ROTsensor is the Rate of Turn as input by an external Rate of Turn Indicator (TI). ROTAIS is rounded to the nearest integer value.
+127 = turning right at more than 5° per 30 s (No TI available)
–127 = turning left at more than 5° per 30 s (No TI available)
–128 (80 hex) indicates no turn information available (default).
ROT data should not be derived from COG information.”

We see that there are two different cases. If the AIS transponder has input from a TI then ROTAIS contains a value in the range from +126 to -126, which covers a range of turning from starboard 708 to port 708 degrees per minute. Sharper turns are clipped, i. e. +-126 is at least 708 degrees per minute. So, the range from -126 to +126 covers all possible turn rates. 126 degrees per minute is slightly less than 0.206 radians per second.

In the other case, without a TI, AIS will not provide a rate of turn value, this is clearly stated in the standard. Instead it will provide one of three values in the form of an enumeration:
    • Turning right (starboard), enum value +127 (0x7f).
    • Heading straight, equal to the default no turn information available, enum value (0x80).
    • Turning left (port), enum value -127 (0x81).
These turns are calculated internally in the transponder based on, I think, heading.

So, ROTAIS is a complex type, a union of a bounded signed integer and a sparsely populated enumeration. Both of these types "shares" the 8-bit binary type of the union. The values of the two types are laid out in none-overlapping regions of this basic 8-bit binary, making it possible for the application to select type based on content of the 8-bit binary.

The rate of turn parameter in PGN 129038, AIS Class A Position Report, field number 14 in canboats pgns.json, is type defined as a simple type, a scaled 16 bit signed integer, with a maximum value of roughly +-1,02 rad/sec. In a sense this is the underlying 16 bit binary type of the union, although delivered in form of a 16 bit signed integer.

Concluding, if the transponder gets rate of turn information from an external sensor, ROTAIS gives a compressed rate of turn value. The value is decompressed in n2k (by the receiving transponder), and scaled to give the value in radians per second rather than degrees per minute. Thus, if the magnitude of the value coming out of analyzer is below 0.206 rad/s, or given limited precision in calculations say 0.207 or 0.208 rad/s, the value given is almost certainly rate of turn, even without knowledge of the enumeration mentioned above. If the magnitude of the parameter is larger than this it is not a rate of turn value. If everything is working correctly it should then be one of the enumerated values. The problem is that the numerical values of the enumerations are not known for two  of them. This was in n2kd/gps_ais handled by changing the value to no turn information available. As it is I would think that should be the path to follow also for other applications.

How harmful is this? A rate of turn indicator is (as far as I know) only required for very large ships, 50000 gross tonnage or more. Likely a gyro-based pretty expensive instrument 20 years ago, most ships with a class A transponder would transmit the enumeration values, not rate of turn. But today there are an ”abundance” of, I think, mems based electronic compasses that provide not only heading, PGN 127250 if on n2k, but also rate of turn, PGN 127251 to a modest price. Consequently it is hard to find the values of the enumeration, and so it will be rare that the application drops an enumeration. Thus it is not a big issue.

But it would be nice to have the two unknown values of the enumeration. If resolution of the union is not done in analyzer, then it can be done in the application.

I scanned candumps passed through analyzer, and found none of the values. Nevertheless, the values should be there in a class A transponder with an n2k port, otherwise it seems to me not fulfilling the standard. So a proposal for how these values may be retrieved. Maybe this can intrigue someone with the equipment to do the work.

Prerequisite is a class A transponder with an n2k port. If it is fed with heading but not rate of turn I think it will output the desired values when the vessel turns. These values can then be caught either with a candump, or by logging the output of analyzer. Filtering for ”VDL transmission” and ”Class A Position Report” should hopefully retrieve the desired reports from own vessel.

The problem is of course to turn off rate of turn but not heading if it comes from the same sensor. Obviously heading sensors (nmea 0183) without rate of turn has been common in the past, so somehow it should be doable.

I think, no guarantees, that the transponder calculates the turn parameter by comparing the current heading with a calculated average heading during the last 30 seconds. If it differs more than 5 degrees it will set the appropriate turning enum value.

So, I think keeping a straight course for a couple of minutes, make a fairy sharp turn of say at least 10 degrees, taking a note of time and direction, keeping the new course for a couple of minutes, and then repeat the operation in the opposite direction should do it.

keesverruijt, 2024-10-05:
Dear Lars,

As I wrote in #162 it is already "fixed" in CANboat. It uses the "unknown" value 0x7fff as in any other N2K datagram for all fields: the maximum (signed) binary value in a field means "not available". For 8 bit signed values this is 0x7f, 8 bit unsigned 0xff, 16 bit unsigned 0xffff etc. Or a boolean value that can be false, true, error or unknown (unset) the bit values are 00, 01, 10, 11.

So: 
- There is no error in CANboat.
- There are no unknown special values in the N2K definition of AIS ROT. 
- There were indeed AIS receivers that incorrectly forgot to treat -128 in AIS ROT specially on their N2K interface. This was a bug in the device firmware, not CANboat. 
- I haven't seen this in years.

To see this in operation, just generate some fake raw messages and feed them to analyzer. Set AIS ROT to 0x7ffff and the output will, depending on the selected output mode, no longer show Rate of Turn, or show it as null.

larsujensen, 2024-10-06:
Dear Kees

We really need to discuss this. Clearly we did not understood each other the last time. And one of us is wrong, I think it is you, you think it is me.

When I opened the issue again it was not out of the blue sky. A couple of weeks ago I saw that you at some time had changed the range check in gps_ais, that I am writing about , from what I firmly believe was a correct check to a principally wrong check, but that highly unlikely will have any effect on the produced output. I intended to write you a separate mail about this, but now it comes here instead.

So I thought, if Kees apparently does not know how to use this parameter, then how on earth are other people supposed to know how to use it? So I tried to write very detailed about the problem and the solution, range check that the rate of turn is within the specified magnitude boundaries 708 degrees per minute.

I have never doubted that canboat treats no turn information available 0x80 (-128) correctly, or rather that the transponder converts the ais 8-bit 0x80 to n2k 16 bit signed integer default value 0x7fff. It is not ais -128 that is the problem, it is +127 and -127. Decompressing these values gives rate of turn with magnitude 720 degrees per minute. That is beyond the 708 max rate of turn value. They are not rate of turn values, but indicates that the ship is turning.

As AIS class A is mandatory for commercial ships above 300 tons, or something like that, I can not possible, possible, possible, imagine that nmea 2000 does not provide all the information from ROTAIS. The information corresponding to ais +127 and -127 must be there if it comes in to the ais. But what are the values on n2k? It could be 720 degrees per minute converted to radians per second, but I find it highly unlikely. Rather something like 0x7ffd and 0x7ffe or, less likely (?) maybe 0x8000 and 0x8001. Or 0x7ffe and 0x8002, I can speculate forever.

So Kees, where are the values corresponding to ais +127 and -127, if they eventually shows up. Because these values are not rate of turn values.

As we write in Sweden
Mvh
Lars

keesverruijt, 2024-10-06:
Aaaahhh, ok, finally an explanation that I understand.

I have no idea what -127 and +127 are translated to. I don't have access to the N2K standard, so I don't know. Your guesses 0x7ffd and 0x7fffe seem the most likely values to me. 

Unlike you I'm not so convinced that NMEA 2000 is perfect. NMEA 2000 is not common on commercial ships as it does not carry wheelmark certification and thus cannot be used for SOLAS applications.

larsujensen, 2024-10-07:
It was meant as a rhetorical question, if you knew we would not have this issue. I assume you understood that.

But I think we are close to a compromise. Maybe we can agree that there might be something lurking in the higher magnitudes of the rate of turn parameter than currently is not present in the specification in canboat? Then as with all fuzzy agreements you do your interpretation, possible but not likely, and I do my interpretation, possible and likely.

Maybe you are right about the wheelmark certification. But there is IEC 61162-3:2008 which is about the application of NMEA 2000 aboard SOLAS vessels. And a quick look shows that e. g. Raymarine AIS4000 claims compliance to IEC 61162-3 2008 Ed 1.2. And isn't wheelmark an European thing only?

I think it should be possible for us to agree that my advice to range check that the value of rate of turn is within the magnitude 708 degrees per minute expressed in radians per second is a good advice. And if not, treat it as no rate of turn available, which is a perfectly valid state.

You brought in yet another dimension to this. We only see the n2k standard through equipment claiming to follow the standard. If they do not, we see false values, and that really depends on individual designers of and testing the equipment.
I thought about how this case with the class B transponder not checking for 0x80 before decompressing the value could come up. Maybe like this. In nmea 0183 the value of ROTAIS is copied as it is to the appropriate sentence. Thus there is no need to check for validity. So, n2k says decompress. The designer takes the value of ROTAIS and decompresses. And you complain. Suppose now that the designer is back to the desk, checking for 0x80 before decompressing. You are happy, but not I! To me the chance that the designer still is not checking 0x7f (+127) and 0x81 (-127) is not insignificant. If we are now looking for how the n2k standard represent these two values we might find 720 degrees per minute expressed in radians per second and say eureka, not knowing that the product is not following the standard.

Hopefully SOLAS certified transponders are tested to comply with the standards they claim to conform to.

TwoCanPlugIn, 2024-10-07:
What an interesting discussion.

Just for the hell of, or perhaps as an exercise in academic masochism, I performed a simple test, generating PGN 1290283 to see how a MFD interpreted the values.

The only conclusion was that the usual N2K signed short values, Data not Available (0x7FFF), Error (0x7FFE) and Not Supported (0x7FFD) resulted in a null field displayed for ROT field on the MFD. For every other value, the MFD displayed mathematically correct ROT values, including values far in excess of the 708 degrees per minute maximum as detailed in the ITU AIS standard. 

I could find no special value that the MFD displayed as "No Turn Indicator and turning faster than +/- 5 degrees per 30 seconds"

> To me the chance that the designer still is not checking 0x7f (+127) and 0x81 (-127) is not insignificant. 

Without access to the NMEA 2000 standard, you are merely postulating. For all we know, the NMEA 2000 standard may define the treatment of AIS ROT values +/- 127 as "Data Not Available"

The only way to verify the interpretation of AIS ROT +/-127 is either via NMEA 2000 standard itself, capturing PGN 129038 upon reception of an AIS type 1,2,3 message known to contain AIS ROT values of +/- 127 or by constructing a NMEA 0183 VDM message, sending to a bi-directional NMEA 0183/NMEA 2000 gateway and see what comes out the other side.

Just because the ITU AIS standard defines a meaning for a ROT value of +/- 127 doesn't imply that either NMEA 0183 or NMEA 2000 have to parse or display the value in a specific meaningful way.

Also as Kees alluded to, vessels that are required under SOLAS to carry Class A AIS transponders must use IMO type approved devices. To the best of my knowledge, no IMO type approved Class A AIS transponders receive input from a NMEA 2000 sensor, nor do I think that any NMEA 2000 devices have IMO type approval.  Even in the case of the unit you mentioned, the Raymarine AIS4000, it specifically states that all sensor data input is from IEC 61162 / NMEA 0183 sentences. 

I'm not going to enter into the discussion about IEC standards, the EU Marine Equipment Directive (Wheelmark) and their certification authorities nor how IMO Type Approvals are issued by certifying authorities.

I can't believe I actually spent some time testing and replying to this.........

larsujensen, 2024-10-08:
First, I feel this has got me out of hand a bit. All I really wanted to advise was to range check the rate of turn parameter, and felt, and still feel, I could motivate that quite well.

Beyond this it is speculations from my side, with limited knowledge in the subject. Anyhow...

Why do you say that "or it may follow the NMEA 0183 standard where +/- 127 indicates a rate of turn to the left or right at 720 degrees per minute (or higher)"? 
Once again this standard is hidden to me, but in https://gpsd.gitlab.io/gpsd/AIVDM.html Eric Raymond specifies the field exactly as in the ais standard. And how on earth shall nmea 0183 be able to produce information that is not there in the over the air protocol? My view is that ROTAIS is copied unengraved to the nmea 0183 sentence. (It might be possible to drop a suitable nmea 0183 sentence into some online tool and see what comes out of it, perhaps. But I am pretty confident in Eric's description.)

And this is really spot on my argumentation. If it is there in nmea 0183, why should the information be removed in nmea 2000? I can not find any sensible motive for it.

larsujensen, 2024-10-09:
No, I don't say that it is removed in nmea 2000. Quite the opposite. I object to canboat assuming it is removed in nmea 2000, which as far as I can see canboat currently is doing. That is the point of this issue. I try to tell users not to make this assumption.

The reason to check boundaries of rate of turn is to guarantee that it is truly a signed integer representing the rate of turn, it has nothing at all to do with limiting allocated storage.

larsujensen, 2024-10-10:
Read the whole standard. It should clarify this apparently weird looking 5 degrees per 30 seconds, and how to interpret it.

But rate of turn is for really big ships, and the parameter is consequently  not present i a class B position report. Do not hold it to me if what i write here is wrong, it is far above my level of navigational certificate.

A rate of turn indicator is a fabulous navigational instrument. When you change heading with the intention to turn, the ship due to inertia will "slide" in the water, i. e. the ship will not go in the same direction as the heading. If you steer the ship at a constant rate of turn at a constant speed it will go in an arc. So you can plan a course change as an arc in the sea chart, and follow this route with the help of the turn indicator. Including the rate of turn value in the position report allows other ships to see this planned route, provided of course that the transmitting ship actually steers with a constant rate of turn.

closed 2025-04-22 by keesverruijt