canboat/canboat #462: PGNS 129038 & 129039 Order 3 missing type

antipole2, 2024-12-14:
I am using the definitions in pgns.json for my own decoding in a plugin for OpenCPN.

IPGNs 129038 & 129039 have
```
"Order": 3,
			"Id": "userId",
			"Name": "User ID",
			"BitLength": 32,
			"BitOffset": 8,
			"BitStart": 0,
			"Resolution": 1,
			"Signed": false,
			"RangeMin": 2000000,
			"RangeMax": 999999999
```
The data type is not specified, although I assume it is Integer.
My processing of this trips up as it does not know how to handle this order.  I can patch my clone of the JSON but therein lies trouble ahead.
Can the type be specified, please?

keesverruijt, 2024-12-21:
Ah. It sounds as if you are using `analyzer/pgns.json` which is an old backwards compatible format.
There are downstream dependencies that use these files and I cannot change these.
You should be using `docs/canboat.json`, this contains either:

          { ...
            "FieldType":"MMSI"
          },

or 

         {
            "RangeMin":0,
            "RangeMax":1099511627773,
            "FieldType":"DECIMAL"
          },

Note that there is still a bug: the RangeMax for DECIMAL is wrong. It should be 9999999999.

I will also try find a way to explain better that `analyzer/pgn.{xml, json}` are obsolete.

Am I correct in assuming you are using the obsolete JSON file?

antipole2, 2024-12-21:
@keesverruijt   You are correct in your assumption.  I have been using `analyzer/pgns.json`.
I take i from you that I should be using `docs/canboat.json`.

I will look into this.  Things there are structured differently, and I may have a bit of work to do to change over.

Thanks for the pointer.

keesverruijt, 2024-12-21:
That is correct, and yes it does require work on your side, and this is why (for now) Canboat still has the old definitions. 
I think I will move them to an `obsolete` directory in 2025 and then remove them completely in 2026.

Anyway, please close the issue if it works for you when you're using `docs/canboat.json`.

antipole2, 2024-12-24:
@keesverruijt  I am working on adapting to this.

I have found that docs/canboat.json line 15946 contains invalid JSON.  The string value for "Explanation" contains multiple unescaped " characters within the quoted string.
I have patched this within my copy but you will want to fix this.

keesverruijt, 2025-01-12:
You may want to read https://github.com/canboat/canboat/wiki/Using-the-definitions-in-other-software

antipole2, 2025-01-14:
Thank you for keeping me informed.
My NMEA2000 constructor now uses docs/canboat.json.
It did take quite a bit of reworking but is usefully simplified as there are fewer data types and more reliance on a consistent way of looking things up.   Definitely an improvement - thank you.