canboat/canboat #463: Fast packet with size 223 (32 frames) doesn't assemble properly

sergei, 2024-12-14:
I'm getting message:

`[analyzer] Received incomplete fast packet PGN 130823 from source 27
`
When trying to decode the attached file with:
`analyzer -format PLAIN_MIX_FAST
`

[00010001-130823-27.log](https://github.com/user-attachments/files/18137443/00010001-130823-27.log)

This debug message revealed the problem:
`Using buffer 0 for reassembly of PGN 130823: size 223 frame 31 sequence 0 idx=216 frames=ffffffff mask=0
`
Clearly mask is not supposed to be a zero

If I change the line:
`p->allFrames = (1 << (1 + (p->size / 7))) - 1;
`
to:
`p->allFrames = (1ULL << (1 + (p->size / 7))) - 1;
`
The problem is gone:
`
Using buffer 0 for reassembly of PGN 130823: size 223 frame 31 sequence 0 idx=216 frames=ffffffff mask=ffffffff
`

keesverruijt, 2024-12-21:
This is a 32 bit only bug.

closed 2024-12-21 by keesverruijt

sergei, 2024-12-22:
Thanks for the fix, although I'm not sure about the title change, can be misleading. My platform is OSX arm64.

commit 04269166 (2024-12-21): Fix #463: Fast packet with size 223 (32 frames) doesn't assemble properly on 32 bit platform