canboat/canboat #468: Unicode characters in canboat.json PGN 127233 field Explanation

keesverruijt, 2024-12-24:
I have found that docs/canboat.json line 15946 contains invalid JSON.  The string value for "Explanation" contains multiple unescaped " characters within the quoted string.
I have patched this within my copy but you will want to fix this.

_Originally posted by @antipole2 in https://github.com/canboat/canboat/issues/462#issuecomment-2560744165_

keesverruijt, 2024-12-24:
@antipole2 the characters that you refer to are "curly" double quotes, not normal ASCII double quotes. So the validator which we run during the build (by reading everything into Python) doesn't see this as breaking any contract, and strictly spoken it is right.

However, I feel that UNICODE characters in the explanation needlessly complicate matters as it requires all downstream users to support UTF-8, so I will remove them.

antipole2, 2024-12-24:
Thanks.
I am parsing using the DukTape JavaScript engine, which only parses ASCII, at least on Windows.

keesverruijt, 2024-12-24:
According to the DukTape documentation it supports UTF-8, but the point stands.

keesverruijt, 2024-12-25:
Some other non-ASCII characters found as well.

Fixing this and adding a test so this cannot regress.

commit d931d115 (2024-12-25): Fix #468: Remove Unicode characters in canboat.json and canboat.xml
closed 2024-12-25 by keesverruijt