canboat/canboat #470: not release all release artifacts are created on release

aldas, 2024-12-25:
https://github.com/canboat/canboat/actions/runs/12495099793/job/34865489144

> Run export CANBOAT_VERSION=$(sed -En 's/.*\ VERSION\ \"([0-9]+\.)([0-9]+\.)?([0-9]+)\"/\1\2\3/p' common/version.h)
Checking if docs/canboat.json contains "CANboat version v5.1.2"
Error: Process completed with exit code 1.

so pgn fields are not added to release.

keesverruijt, 2024-12-26:
Good catch! I didn't even notice. Updated my workflow slightly so this is not likely to occur again. Fixed in v5.1.3.

closed 2024-12-26 by keesverruijt

aldas, 2024-12-27:
@keesverruijt I think there is setting somewhere that will make Github to email you when actions fail.  I think this
https://docs.github.com/en/account-and-profile/managing-subscriptions-and-notifications-on-github/setting-up-notifications/configuring-notifications#github-actions-notification-options

I use this for my own repos/maintained projects