canboat/canboat #471: RANGE_RESIDUAL_MODE names need reconsidering

antipole2, 2024-12-27:
In canboat.json lines 2229 - 2226 is
`      {
        "Name":"RANGE_RESIDUAL_MODE",
        "MaxValue":3,
        "EnumValues":[
          {"Name":"Range residuals were used to calculate data", "Value":0},
          {"Name":"Range residuals were calculated after the position", "Value":1}
        ]
      },`

I doubt that this is satisfactory. My sample data has a value of 3 which has no name.

ChatGPT offers the following.  While I do not trust ChatGPT as always gospel truth, it may be a better guide.

> In NMEA2000 PGN 129540, what are the range residuals mode and what values can it take?

In NMEA2000 **PGN 129540 (GNSS Sats in View)**, the **"Range Residuals Mode"** field indicates the method or type of calculation used for determining the range residuals. This field provides additional context for interpreting the residual values reported for each satellite.

### Details of Range Residuals Mode:

1. **Definition:**
   - The **Range Residuals Mode** specifies the basis or source of the range residuals measurement. This helps users understand how the residuals were derived.

2. **Possible Values:**
   - The exact values and their meanings are defined in the NMEA2000 standard and may vary slightly depending on the implementation. Typically, the values are represented as a numeric code, such as:
     - **0**: Not available (the mode of residuals calculation is not specified or cannot be determined).
     - **1**: Based on pseudorange residuals (derived directly from the raw pseudorange measurement).
     - **2**: Based on carrier phase residuals (calculated using carrier phase measurements, often more precise).
     - **3**: Based on both pseudorange and carrier phase (a combined calculation for improved accuracy).
     - **4**: Other (a non-standard or proprietary method is used for calculating residuals).

3. **Significance of Each Mode:**
   - **Pseudorange residuals** are generally less precise but are widely used in standard GNSS positioning.
   - **Carrier phase residuals** provide higher accuracy but require more sophisticated processing.
   - The combined mode offers a balance between accuracy and processing complexity.

4. **Use in Applications:**
   - Understanding the range residuals mode is important for evaluating the quality and reliability of the GNSS signal.
   - Systems requiring high precision (e.g., differential GNSS or RTK) may rely on carrier phase residuals.
   - For general navigation, pseudorange residuals are usually sufficient.

5. **Interpreting the Value:**
   - If the mode is "Not available" (0), the residuals reported may not provide meaningful information.
   - When a specific mode is indicated, it allows the user or system to apply appropriate corrections or weighting to the residual data.

keesverruijt, 2025-01-06:
The fact that it describes value 0 as meaning "Not available (the mode of residuals calculation is not specified or cannot be determined)" suggests this is incorrect, as such values in NMEA2000 are generally represented by an all-ones value. So your "3" has no name, as it is, according to our current database, the value for "unknown/invalid". 

If I ask Copilot, it gives me the same answer but with references:
https://github.com/canboat/canboat/discussions/344
https://www.mavyn.com/blog/understanding-pgn-129540-a-detailed-guide

The first doesn't say that the values are what the AIs say, besides it is a discussion in this same repo, and the 2nd doesn't even say anything.

Now it _could_ be that the NMEA has lengthened the range residual mode field by one or more bits, but then the reserved field should show up with an unexpected value.

I checked but the latest 129540 sample data from reputable GNSS sources do not support your argument.

Can you please show some raw (unanalysed) data?

antipole2, 2025-01-06:
Sorry - unable to supply sample data.
My data is as passed on from OpenCPN and I do not have access to the raw Can data.  Also, I am wintering in the UK while my boat winters near Copenhagen - so no chance of getting any.

I accept your point that if the field really is only two bits long and hence 3 bits on indicates unavailable or undefined.  So the explanation I quoted earlier must be wrong.

I put this to ChatGPT and it has had second thoughts!

> But in NMEA2000, all bits on is used to inidicate undefined or not available.  so how come you are saying 3 (all 2 bits on) is something else?

ChatGPT said:

> ChatGPT
> You're absolutely right—according to the NMEA 2000 standard, all bits set to 1 in a field typically indicate an undefined or unavailable value. Let's clarify the situation for the Range Residual Mode in PGN 129540:
> 
> Range Residual Mode is a 2-bit field.
> The values assigned are as follows:
> 00 (binary) = Not available (undefined)
> 01 (binary) = Differential correction applied
> 10 (binary) = Reserved
> 11 (binary) = Not available (undefined)
> This aligns with the NMEA convention where 11 (all bits set to 1 for a 2-bit field) is used to indicate that the value is undefined or unavailable. Therefore, my earlier suggestion that 11 (decimal 3) had a specific meaning was incorrect.
> 
> Thanks for catching that! Let me know if you'd like further clarification.

So a vlaue of 01 'Differential correction applied' tallies with canboat's "Range residuals were calculated after the position"

So happy to mark this discussion as erroneous and closed.

Thanks for your attention.

aldas, 2025-01-06:
For me ChatGPT does not provide `4: Other (a non-standard or proprietary method is used for calculating residuals)` as value.  (I am using o1-mini model).

These "AI"s are very dubious sources for information.  Probably everyone knows that pretty much parrot info that was written somewhere by someone, maybe even similar discussion where people guessed what these fields MAY contain.

keesverruijt, 2025-01-06:
I tried some other questions in Copilot, as that seems to be most happy to divulge where it gets the data from, and I can't really see that it has real insight into what it is telling me. The answer ChatGPT gave to you in the second instance is still wrong!

Closing for now. If once you're on board again you find that your GPS is sending out a "not all ones" reserved field, indicating this field was lengthened, create a new issue.

closed 2025-01-06 by keesverruijt