canboat/canboat #475: JSON FieldTypes inconsistencies

meaning-matters, 2025-01-07:
In the FieldTypes JSON I'm seeing several parameters that are present for only a few entries. I find that strange because in my view FieldTypes should be able to drive generic code logic that takes a fixed set of FieldType parameters and is able to do the proper field conversion with it. Having several parameters that apply for only a few FieldTypes means that the current FieldType design is unable to generically define all field conversions.

• TIME and DATE appear in both PhysicalQuantities and FieldTypes. Clearly PhysicalQuantities is the proper place to define units and FieldTypes the field format. Therefore TIME and DATE in FieldTypes having a Unit seems wrong/useless.

• The FieldTypes parameter VariableSize appears only twice and are true in both cases. Why don't the other entries have this parameter? Why doesn't even VARIABLE have this parameter, or BINARY? Much better to get rid of this parameter and select the proper conversion in code based on Name: STRING_LZ or STRING_LAU.

• The signed parameter is even more confusing. DATE has signed (why?), but TIME does not. NUMBER does not have signed, but MMSI does, ... All very weird and the JSON clearly doesn't define a full consistent format; all kinds of addition information is needed to know what it means to have the signed parameter or not.

keesverruijt, 2025-01-08:
If you want a simple but complete converter you should be able to write a complete converter of any kind without using PhysicalQuantities and FieldTypes. The PGN fields should contain everything you need. The reason they are present is to generate the full documentation and to validate the definitions. However, now that they are in there, removing stuff what you call "complexity" is no longer possible, as I can't remove any part of the XML or JSON without potentially breaking downstream code.

I will work to remove as much of the inconsistencies that you noted as possible.

meaning-matters, 2025-01-09:
Isn't FieldType needed to parse a raw PGN and get proper field values? Also, without knowing 'the nature' (i.e. unit/physical-quantity) of the values, how useful are parsed field values? I'm therefore confused what you could mean with 'a complete converter'. (I want to build a PGN parser and display field values in a UI.)

keesverruijt, 2025-01-12:
Nah, take a typical value like this like the water depth in PGN 128267. For your benefit I'll show the JSON version:

          {
            "Order":2,
            "Id":"depth",
            "Name":"Depth",
            "Description":"Depth below transducer",
            "BitLength":32,
            "BitOffset":8,
            "BitStart":0,
            "Unit":"m",
            "Resolution":0.01,
            "Signed":false,
            "RangeMin":0,
            "RangeMax":42949672.93,
            "FieldType":"NUMBER",
            "PhysicalQuantity":"LENGTH"
          },

The minimal fields that are required to create a tool that decodes the data and present it to a human are Name, BitLength, Unit, Resolution, Signed and FieldType. The Range values, for instance, are just a result of the fact that the field is an unsigned 32 bit number, with each binary increment contributing 0.01 m. Unsigned ("Signed") 32 ("BitLength") number ("FieldType") with each binary increment contributing 0.01 ("Resolution") meters ("Unit").

But you do not need the NUMBER field type which doesn't add any value other than to you as a developer:

    {
        "Name":"NUMBER",
      "Description":"Number",
      "EncodingDescription":"Binary numbers are little endian. Number fields that use two or three bits use one special encoding, for the maximum value.  When present, this means that the field is not present. Number fields that use four bits or more use two special encodings. The maximum positive value means that the field is not present. The maximum positive value minus 1 means that the field has an error. For instance, a broken sensor. For signed numbers the maximum values are the maximum positive value and that minus 1, not the all-ones bit encoding which is the maximum negative value.",
      "URL":"https://en.wikipedia.org/wiki/Binary_number"
      },

nor do you need to explain to a user what LENGTH means ("PhysicalQuantity") unless you want to.

keesverruijt, 2025-01-12:
And on this lazy Sunday afternoon I finally have some time to go into your other points.

First of all, as a JSON user I think you skipped `docs/canboat.xsd` which defines the schema in `docs/canboat.xml`, and thus it contains a lot of material in the form of comments that explain what the fields in the XML (and JSON) mean and when you can expect them to be present.

> TIME and DATE appear in both PhysicalQuantities and FieldTypes. Clearly PhysicalQuantities is the proper place to define units and FieldTypes the place for field formats. Therefore TIME and DATE in FieldTypes having a Unit seems wrong/useless. Decoding logic needs to look at the PhysicalQuantities for unit, so these two units in FieldType should be deleted.

As you can read in the XSD document:

      <xs:element type="xs:string" name="FieldType">
        <xs:annotation>
          <xs:documentation>
            Fields always contain a field type.
          </xs:documentation>
        </xs:annotation>
      </xs:element>

So there has to be a field type, and thus one for TIME and DATE fields as you cannot use the same code to display them as a number in anything that wants to be human friendly.

> The FieldTypes parameter VariableSize appears only twice and are true in both cases. Why don't the other entries have this parameter? Why doesn't even VARIABLE have this parameter, or BINARY? Much [...]
Because all DATE fields are unsigned, but TIME fields occur in signed and unsigned forms.

> NUMBER does not have Signed, but MMSI does, ... 
Because MMSI is always unsigned, so the Fieldtype contains "Signed = false", but for NUMBER this again depends on the actual PGN field, some are signed and some unsigned, so NUMBER cannot make any assertion about the signedness.

> The JSON doesn't seem to define a full consistent format; all kinds of additional information is needed to know what it means to have the Signed parameter or not. 
That's the fault of JSON not having a proper schema like XML.
> So the Signed parameter could as well be deleted from JSON and be noted in the comment; the code can simply do the proper conversion based on Name.

There is reason to the madness, see below.

> 
> • The Bits parameter, although seemingly cleaner than the above three, could also be deleted from the JSON.
> 
> Instead of having various optional parameters which results in all kinds of if-else logic in the code, I think it's much better to accept the fact that there are very different FieldTypes for which it's impossible to define a clean/simple set of parameters to properly steer conversion code logic. Just remove these optional parameters and allow code logic to switch to the proper conversion based on Name only. With the small number of FieldTypes there's no need for all this complexity in the JSON.

Again I just refer to the XSD document, which makes it pretty clear what I intended:

      <xs:element type="xs:unsignedShort" name="Bits" minOccurs="0">
        <xs:annotation>
          <xs:documentation>
            The number of bits that this field takes up in the PGN, if this
            is fixed for all fields.
          </xs:documentation>
        </xs:annotation>
      </xs:element>
      <xs:element type="xs:string" name="Unit" minOccurs="0">
        <xs:annotation>
          <xs:documentation>
            Present in the FieldType when it has a typical unit.
          </xs:documentation>
        </xs:annotation>
      </xs:element>
      <xs:element type="xs:boolean" name="Signed" minOccurs="0">
        <xs:annotation>
          <xs:documentation>
            Present in the FieldType when all fields have a garuanteed sign.
          </xs:documentation>
        </xs:annotation>
      </xs:element>
      <xs:element type="xs:boolean" name="VariableSize" minOccurs="0">
        <xs:annotation>
          <xs:documentation>
            True if the fields of this FieldType all have a variable length in bits.
          </xs:documentation>
        </xs:annotation>
      </xs:element>

keesverruijt, 2025-01-12:
See https://github.com/canboat/canboat/issues/462

You should be using the files in the `docs/` directory.