canboat/canboat #476: EnumFieldTypeValues have lowercase `name' parameter

meaning-matters, 2025-01-07:
All structures, except EnumFieldTypeValues, in the JSON have a capitalized "Name". In  EnumFieldTypeValues it's "name".

Seems to be an oversight. Easy to fix.

keesverruijt, 2025-01-08:
> All structures, except EnumFieldTypeValues, in the JSON have a capitalized "Name". In EnumFieldTypeValues it's "name"; also "value" instead of "Value".
> 
> Seems to be an oversight. Easy to fix.

Easy to fix, but I will have to change the semver version of the definitions as downstream projects might depend on it. 
I prefer to do postpone this until I have some other changes, as I like to keep the definition structure stable.

> 
> Then, why are PhysicalQuantitities not used in EnumFieldTypeValues, but instead loose Units?

There is no good reason, other than that the entire PhysicalQuantities definition was deemed not necessary by other downstream developers as "needless complication." I like it and I've sneaked it back in, but not fully. The attributes FieldType, Signed, Resolution, Unit, Bits belong at LookupFieldTypeEnumeration not EnumFieldType. 

Changing it to move up is already the 2nd change to the structure :-)

> 
> Also, "Bits" are string here, while everywhere else in the JSON "Bits" is an integer.

3rd change :-)

> 
> (Why is "Anchor depth" in "m/s"?)

Bug! (for myself later: this in SIMNET_KEY_VALUE)

meaning-matters, 2025-01-09:
You're right, changing the JSON/XML format has a big impact.

It's easy for me to work with these few 'issues' in my JSON parsing.

Then, as I replied elsewhere: Don't understand how people could do without PhysicalQuantities; where do they get the Unit and related information from then?

aldas, 2025-01-09:
> Then, as I replied elsewhere: Don't understand how people could do without PhysicalQuantities; where do they get the Unit and related information from then?

depending on the application that you are building. I for example just store raw values to database for debugging reasons.  for such use-cases pretty much only FieldTypes are things that you care about.

> You're right, changing the JSON/XML format has a big impact.

as one of the down-stream users of canboat - schema changes are not a very big problem when they are documented somewhere.  You can always lock you version to some canboat release (I use release page artifacts for that) and main problem is to understand what has changed (and how) when time comes to upgrade your application code to support newer canboat schema.

keesverruijt, 2025-01-12:
Oh, daft, as I am now working on finding your changes, I realise that you have been looking at the OBSOLETE `analyzer/pgns.json` file. This is the 2nd time in a month that someone picks these old files.

Blast. Time to deprecate these old files into a deprecated directory.

keesverruijt, 2025-01-12:
I wrote a new Wiki page: https://github.com/canboat/canboat/wiki/Using-the-definitions-in-other-software

meaning-matters, 2025-01-12:
@keesverruijt I've only been looking at `docs/canboat.json` and have been parsing only that file.