canboat/canboat #477: Typo in RangeMax for Oil temperature in PGN 127489. RangeMin for temperatures should be at least 1

negrusti, 2025-01-08:
RangeMax for Oil Temperarure should be 655.33, not 6553.3
```
        "PGN":127489,
        "Id":"engineParametersDynamic",
        "Description":"Engine Parameters, Dynamic",
        "Priority":2,
        "Type":"Fast",
        "Complete":true,
        "FieldCount":14,
        "Length":26,
        "TransmissionInterval":500,
        "Fields":[
          {
            "Order":1,
            "Id":"instance",
            "Name":"Instance",
            "BitLength":8,
            "BitOffset":0,
            "BitStart":0,
            "Resolution":1,
            "Signed":false,
            "RangeMin":0,
            "RangeMax":253,
            "FieldType":"LOOKUP",
            "LookupEnumeration":"ENGINE_INSTANCE"
          },
          {
            "Order":2,
            "Id":"oilPressure",
            "Name":"Oil pressure",
            "BitLength":16,
            "BitOffset":8,
            "BitStart":0,
            "Unit":"Pa",
            "Resolution":100,
            "Signed":false,
            "RangeMin":0,
            "RangeMax":6553300,
            "FieldType":"NUMBER",
            "PhysicalQuantity":"PRESSURE"
          },
          {
            "Order":3,
            "Id":"oilTemperature",
            "Name":"Oil temperature",
            "BitLength":16,
            "BitOffset":24,
            "BitStart":0,
            "Unit":"K",
            "Resolution":0.1,
            "Signed":false,
            "RangeMin":0,
            "RangeMax":6553.3,
            "FieldType":"NUMBER",
            "PhysicalQuantity":"TEMPERATURE"
          },
```

keesverruijt, 2025-01-08:
No, oilTemperature is okay this way; it is defined with 0.1 K resolution, not 0.01 like other temperatures. Apparently the committee felt that oil temp can reach more than 655 - 273 = 382 degrees C.

keesverruijt, 2025-01-08:
I also don't get why you say that RangeMin should be 1, apart from the unlikeliness that an on-board temperature will be zero Kelvin...

negrusti, 2025-01-09:
@keesverruijt got it re. Oil temp. 
Regarding the RangeMin - setting it to a realistic number will allow automatic filtering of default (invalid) values - as it works with Radians for example. The software can implement it's own filtering but I believe it is much better to set it in the definitions. What's the downside of changing it to a positive number?

negrusti, 2025-01-09:
[image]
An example - this engine sensor is sending the coolant temperature but not the oil temperature - with proper RangeMin I can filter Oil Temperature entry automatically, instead of adding a special case handler.

keesverruijt, 2025-01-09:
> On 9 Jan 2025, at 09:39, Grigory Morozov ***@***.***> wrote:
> Regarding the RangeMin - setting it to a realistic number will allow automatic filtering of default (invalid) values - as it works with Radians for example. The software can implement its own filtering but I believe it is much better to set it in the definitions. What's the downside of changing it to a positive number?
> 
Well, first of all, the current system explains what values you can theoretically get from a device, not what would be a (arbitrary) feeling on what is reasonable.

What is a "reasonable" number for minimum (oil) temperature? -50 C? -200 C?

Note that in NMEA 2000 any properly programmed sensor will use the ERROR or UNKNOWN value. In case of Oil Temp, which is an unsigned 16 bit integer, these are the values 65534 and 65535 respectively. I've misplaced where that is documented, I will follow up on that.

keesverruijt, 2025-01-09:
If your device is sending out 0 (in bits) for an unknown oil temperature it is not following NMEA 2000 specifications and this should be reported to the manufacturer. It should be "all ones".

keesverruijt, 2025-01-09:
I'd be curious to know what device that actually is... Brand, type and sw revision?

negrusti, 2025-01-09:
> I'd be curious to know what device that actually is... Brand, type and sw revision?

Unfortunately I don't know - I am testing my own software on other people's logs

closed 2025-02-09 by keesverruijt