canboat/canboat #478: Make analyzer/pgn.xml+json obsolete

keesverruijt, 2025-01-12:
Too many new devs turn up and find `pgns.xml` and `pgns.json`. They should be made more obviously obsolete, the `docs/` files should be used instead.

keesverruijt, 2025-01-12:
Gents, it's time to move off of the v1 files `pgns.xml` and `pgns.json`, use `docs/canboat.{xml, json}` instead.
So far, this branch moves them from `analyzer/` to `obsolete/` and nothing else here in canboat.
I'm happy to keep the v1 generation intact here, but it really needs to be obvious to new developers coming to canboat that they are not to be used in new software. There were two new devs that fell for this in the last months.

@sbender9 

I think we can merge this and then separately create a @canboat/pgns2 NPM, or just update @canboat/pgns to use canboat.json instead? Then update @canboatjs and warn the other two packages dependent on it that they need to update their code as well?

@mairas 

I think we can just leave the abc-exporter use these as they don't really do much with all the complicated stuff anyway? I had a look and the JSON reader will have to create a Lookup object and then refer to those in the Fields, if we want to move it off the obsolete files.

@tkurki 

Any thoughts?

sbender9, 2025-01-12:
I am already using canboat.json. I don't need the old stuff at all. As far as I'm concerned, you can delete them.

keesverruijt, 2025-01-12:
> I am already using canboat.json. I don't need the old stuff at all. As far as I'm concerned, you can delete them.
> 
> Nothing needs done on the canboatjs side.

In NPM the @canboat/canboatjs package claims to be dependent on @canboat/pgns which is the old files?

keesverruijt, 2025-01-12:
Certainly `package.json` in `analyzer/` refers to the old files.

sbender9, 2025-01-12:
I don't actually use that anymore. You can remove it.

sbender9, 2025-01-12:
And we don't need pgns2, the existing pgns package contains the old and new files.

sbender9, 2025-01-12:
https://github.com/sbender9/pgns

I manage that package manually because I have tests that need to be run in multiple packages before I release a new version.

aldas, 2025-01-14:
@keesverruijt , why do not you delete things that are obsolete - you are making things harder for yourself in maintenance perspective?  

You could just state in README.md that linking to master branch is discouraged and could result to compile time etc breaking changes and suggest people to pin/lock their software to Canboat release artifacts or git release tags.

commit 59e42b79 (2025-05-04): Fix #478
closed 2025-05-04 by keesverruijt