canboat/canboat #480: PGN 65005 units is incorrect

bobeast, 2025-01-16:
At least in Maretron's usage, PGN 65005 expresses values in Joules. It is listed as expressing kWh.

TwoCanPlugIn, 2025-01-23:
From a copy of the SAE standard that I found at this [link](https://d1.amobbs.com/bbs_upload782111/files_23/ourdev_515235.pdf) it seems to be in kwH

bobeast, 2025-01-24:
Interesting. Perhaps Maretron is not following that spec correctly, as they are outputting values in joules. Either that or signalK may be intercepting and changing it to joules. I have not looked at the raw sentence.

TwoCanPlugIn, 2025-01-24:
Indeed a quandary.

Joules are SI units whereas kwH are not. Both J1939 and NMEA 2000 use SI units. Either the standard document I provided a link to is out of date or incorrect, and/or Maretron & SignalK are interpreting or converting to SI units.

Really need some live data with a known value to verify.

sbender9, 2025-01-24:
Signal K is not doing any conversion for the values in that pgn.

keesverruijt, 2025-01-25:
I think this range of PGNs (below 0x10000, e.g. ) is somewhat special and was copied/derived from J1939-75. They're not in the PGN list of NMEA 2000 v1.300 so I assume they were added between v1.300 and v2.000. 

Of course J1939 has the same problem: it is not an open standard.

Searching the internet I find various, mostly really old, copies of J1939-75 and newer versions for sale.

I'll go and check other J1939 derived quantities, but it seems that PGN 65005 is correct as the J1939 standard says:

GENERATOR TOTAL KW HOURS EXPORT—The total kilowatt-hours that have been exported by
the generator.
Data Length: 4 bytes
Resolution: 1 kWh/bit gain, 0 offset
Data Range: 0 to 4211081215 kWh
Type: Measured
Suspect Parameter Number: 2468

That's for a different PGN (65018) but it seems J1939 has no qualms specifying data in non-ISO units. 

We could of course just do the conversion in CANboat, e.g. say that the data is in Joules with 1 bit value = 3600000 J. But I think this is not the right track in CANBoat itself: it should reflect the data in the PGNs, whatever it is. E.g. ISO units for NMEA 2000 original PGNs and whatever it is for J1939.

Not sure what to do downstream in Signal K, where the community does have full authority to decide what the units are. I think kWh is pretty acceptable as it is a secondary derived unit but real SI (Joule) would be more consistent.

keesverruijt, 2025-01-28:
Hmm, I tried creating an artificial `.ebl` file and letting Actisense EBL reader parse a PGN-65005 but they chicken out and it just says 'J1939 catch all' without any field explanation.

We'll need someone with a Maretron device outputting this I think before we can say what is what.

Note that at the moment, although violating the Signal K principle that data is always in SI units, no heuristics are needed: the output is definitely claiming that the output for this field is in kWh. Whether that is correct though remains the question. And if we add units here, shouldn't they be everywhere? That's a big change and a considerable overhead.

sentientapp, 2026-04-28:
My Maretron ACM100 uses kWh for PGN 65005.

```
2026-04-18T04:53:05.272Z 3 193 255  65005 Utility Total AC Energy:  Total Energy Export = 45182 kWh; Total Energy Import = 0 kWh
2026-04-18T04:53:05.272Z,3,65005,193,255,8,7e,b0,00,00,00,00,00,00
2026-04-18T05:17:30.281Z 3 193 255  65005 Utility Total AC Energy:  Total Energy Export = 45183 kWh; Total Energy Import = 0 kWh
2026-04-18T05:17:30.281Z,3,65005,193,255,8,7f,b0,00,00,00,00,00,00
2026-04-18T05:44:55.309Z 3 193 255  65005 Utility Total AC Energy:  Total Energy Export = 45184 kWh; Total Energy Import = 0 kWh
2026-04-18T05:44:55.309Z,3,65005,193,255,8,80,b0,00,00,00,00,00,00
2026-04-18T06:14:40.321Z 3 193 255  65005 Utility Total AC Energy:  Total Energy Export = 45185 kWh; Total Energy Import = 0 kWh
2026-04-18T06:14:40.321Z,3,65005,193,255,8,81,b0,00,00,00,00,00,00
2026-04-18T06:45:05.332Z 3 193 255  65005 Utility Total AC Energy:  Total Energy Export = 45186 kWh; Total Energy Import = 0 kWh
2026-04-18T06:45:05.332Z,3,65005,193,255,8,82,b0,00,00,00,00,00,00
```
---

Page 32 of the [ACM100 User's Manual](https://web.archive.org/web/20230512042813/https://www.maretron.com/support/manuals/ACM100UM_1.3.pdf) specifies that the units are kWh:

[image]

---
Signal K displays the value in Joules:

[image]

---

Let me know if I can provide any additional data to close this issue.