canboat/canboat #483: PGN 127506

c-logic, 2025-02-05:
0x1F212: PGN 127506 - DC Detailed Status

Field#7	Ripple Voltage	0.001 V	0 .. 655.33	16 bits unsigned NUMBER

instead of

Field#7	Ripple Voltage	0.01 V	0 .. 655.33	16 bits unsigned NUMBER

keesverruijt, 2025-02-05:
You can't increase the precision by one digit without changing the range, I presume you missed that? 

Do you have any public source or screenshots you can share that corroborate this change?

PS I think it's quite likely that a ripple voltage has a higher precision and a lower range, but I would like to base this on something besides you telling me to.

c-logic, 2025-02-08:
refered by Zeus3s

The Value was 1000 (int16)

[image]

keesverruijt, 2025-02-09:
Super. thanks.

commit 2ce58ac5 (2025-02-09): Fix #483
closed 2025-02-09 by keesverruijt