canboat/canboat #492: Fallback PGNs are missing from canbaot.json

tomer-w, 2025-03-20:
[My](https://github.com/tomer-w/nmea2000) python NMEA 2000 decoder which is based on canboat failed some tests. Apparently, as my data source is canboat.json, it is missing all the Fallback PGNs.

From canboat2json.xslt:
```xslt
<xsl:template match="PGNs">
    "PGNs":[
      <xsl:for-each select="*">
        <xsl:choose>
          <xsl:when test="./Fallback = 'true'">
          </xsl:when>
          <xsl:otherwise>
            <xsl:apply-templates/>
            <xsl:if test="following-sibling::*">,</xsl:if>
          </xsl:otherwise>
        </xsl:choose>
      </xsl:for-each>
    ]
  }</xsl:template>

```
I find it strange that ```./Fallback``` is filtered out. It means that my source of truth is partial. What is the reason to filter it out? Can I submit a fix to remove this choose clause?

keesverruijt, 2025-03-21:
> I’m fine with removing this filter, but as a separate PR not mixed in with your other quest. 
> 

I’m happy to remove it myself as well, without a PR.

tomer-w, 2025-03-21:
PR submitted: https://github.com/canboat/canboat/pull/493

keesverruijt, 2025-04-06:
Merged.

closed 2025-04-06 by keesverruijt