canboat/canboat #496: CanBoat.xml improvements in NAME fields

Elmue, 2025-04-05:
You did a huge work in creating the XML file.
However there are some improvements possible.

In PGN 126983 there are two NAME fields of **64 bit** length. Also other PGNs contain NAME fields.

You declare these as:

<RangeMin>0</RangeMin>
<RangeMax>18446744073709551615</RangeMax>
<FieldType>NUMBER</FieldType>

It is surely not a clever idea to display such a huge number to the user which gives not useful information at all.

The NAME field is described in detail in SAE J1939-81
It consists of
- Identity Number (SPN 2837)  BitOffset = 0, BitLength = 21
- Manufacturer code (SPN 2838)  BitOffset = 21, BitLength = 11
- Controller ID (SPN 2554)  BitOffset = 32, BitLength = 32  (may be displayed as HEX or may be omitted as it is double information)
- ECU instance (SPN 2840)  BitOffset = 32, BitLength = 3
- Function Instance (SPN 2839)  BitOffset = 35, BitLength = 5
- Function (SPN 2841)  BitOffset = 40, BitLength = 8
- Reserved BitOffset = 48, BitLength = 1
- Vehicle System (SPN 2842)  BitOffset = 49, BitLength = 7
- Vehicle System Instance (SPN 2843)  BitOffset = 56, BitLength = 4
- Industry Group (SPN 2846)  BitOffset = 60, BitLength = 3
- Arbitrary Address Capable (SPN 2844)  BitOffset = 63, BitLength = 1

Why don't you replace the 64 bit NAME field with these fields in XML ?

They require enums for:
- Manufacturer code  (see https://www.isobus.net/isobus/manufacturerCode)
- Industry Group  (see https://www.isobus.net/isobus/nameFunction/index)
- Vehicle System  (see https://www.isobus.net/isobus/nameFunction/index)
- Function  (see https://www.isobus.net/isobus/nameFunction/index)

Instead of a useless 64 bit number the XML file would decode this huge number into a human understandable text string.

keesverruijt, 2025-04-07:
Please, can you create separate issues for separate items? This is not a chat but an issue tracker.

keesverruijt, 2025-04-07:
NAME is a single field because it is a single field in the NMEA 2000 PGN description, and in most cases the meaning of the field is not that interesting. Also, changing XML field names and changing their order/# is a huge burden on downstream packages. I will not be doing that for this case. What we _can_ do is make it a new type, and introduce a nested structure definition for those that are interested.

> It would also be useful if you introduce a new <FieldType>PGN</FieldType> which allows to display not only the PGN number to the user but also the meaning of this number.

Adding something akin for PGN would be doable as well.

> Another improvement would be to replace cryptic abbreviations in XML with names that everybody will understand.
For example "SID" with "Sequence ID", "COG" with "Course over Ground" and "SOG" with "Speed over Ground", "DOP" with "Dilution of Precision".

What you call "cryptic" is SOP in the industry, I think a lot of non-native (and native) English speakers are quite used to SOG, COG. Also, changing field names is not in scope for frivolous reasons as all down stream projects are affected.

> It would also be a good idea to change the formulas so angles are calculated in degree instead of radians and temperatures in Celsius instead of Kelvin which are much more human understandable.

NMEA 2000 uses ISO units for its representation, so the XML describes it in these terms as well. End user tools should obviously convert to user-practical representation, but that is out of scope for the XML and JSON files. Note that `analyzer` does do these conversions (but is generally not used like that, it is a developer's tool not an end user.)

> It would be useful if the XML file would make a difference between times and intervals.
You declare both as <FieldType>TIME</FieldType>
One should be displayed a clock time with AM / PM the other one as an interval in seconds, minutes, hours and days.
It is currently not possible to distinguish them.
For example "Total Engine Hours" may be more than 24 hours on a vessel that makes a long journey, while the clock time restarts from zero after 24 hours. So these must be distinguished for a correctly formatted display.
Additionally intervals may be negative (TCPA in PGN 128520) while clock time is never negative.

Well, AM/PM wall clock presentation is a relic presentation in use in some geographies only (usually they also use feet instead of meters, and Fahrenheit as opposed to meters), most of the world uses a 24 hour clock, so for them there is no difference in presentation. But you are right, `TIME` is usually an interval -- I think there are only a few cases where it represents a wall clock time. Generally most computer time packages call this a duration, which is probably an even better word. I will add a DURATION field type in the next minor version (coming up soon, because of issue #486).

Elmue, 2025-04-07:
> Please, can you create separate issues for separate items?

No problem.
I didn't want to flood you with lots of issues.
I thought it would be no problem for you to leave an issue open until all tasks inside are finished.

> Well, AM/PM wall clock presentation is a relic presentation in use in some geographies only

Correct.
I offer my users the option to chose their time format as they like.
For the users from USA I show times as hours from 0 to 12 with AM / PM.
European users will not use this option.

However I need to distinguish what is time and what is interval.
Currently the only way is to parse the Description and search for the word "midnight", which is an ugly workaround.
`<Description>Seconds since midnight</Description>`
So if you introduce the type DURATION in addition to TIME this would be perfect.

Does Gihtub inform you about comments in issues that are already closed ?
I added an answer to issue 500

By the way:
Thank you for the great work that you are doing in this project !
Without CanBoat implementing NMEA would be an endless or even impossible work.

closed 2025-04-22 by keesverruijt