canboat/canboat #498: Contradicting information in CanBoat.xml

Elmue, 2025-04-06:
You write for STRING_LZ:
"A varying length string containing single byte codepoints encoded with a length byte and terminating zero.
The length of the string is determined by a starting length byte."

But then you define a fix BitLength for all STRING_LZ as 40 or 80 or 96 or 104 or 256 bits, 

This is contradicting.
Please make clear if STRING_LZ is either fix length or variable length. 
It cannot be both at the same time.

keesverruijt, 2025-04-07:
For most of these fields we think we have seen practical limits on the _maximum_ length of the STRING_LZ field. I think that is what it says for all these PGNs: we don't really know the maximum length. A safe implementation will allow for any length of (up to 233 bytes if a fast PGN is involved). For instance, when I reverse engineered the SonicHub PGNs (now the Fusion and generic audio PGNs) it seemed there was a maximum track and album name length much shorter than 233. 

For STRING_LAU I have more hints that the length is indeed unspecified and that the maximum is bounded by the max PGN size.

So what I will do is add a note to STRING_LZ is that the field length is just an observation in the field, that may be wrong.

Elmue, 2025-04-07:
Very inconsistent these guys from Fusion!
They introduce a length byte and then they don't use it ?
So how do I handle these LZ strings ?

I see that in all messages this type of string is the last signal in the message.
So I simply read to the end of the message and skip the preceding length byte ?

How do the strings look in the real world?
Do they have a fix length and they fill up the remaining bytes with zeros ?
Which logfile do you recommend me to study the data ?

keesverruijt, 2025-04-12:
TBH these are only defined for SonicHub (last produced in 2014 I think) and Fusion specific. They have been made obsolescent by the generic audio PGNs.

Maybe somebody can create a logfile.

I will update the doc to point out that we don't know.

tomer-w, 2025-04-22:
@keesverruijt , yes, this is my understanding based on their code and the new sample file.

TwoCanPlugIn, 2025-04-23:
I think it is fair to say that Fusion are an anomaly. (Their use of a proprietary Fast Message, PGN 126720 to send commands with less than 8 bytes in length attests to that!)

Here are samples I collected when reverse engineering PGN 130820 transmitted from a Fusion Media Player. 

Even though they contain a string length, they also include a trailing null character, which is not included in the string length calculation. So I think the current definition of STRING_LZ could be clarified and also clearly identified as only observed on Fusion devices.

This is quite different form other encodings within NMEA 2000, for example what you describe as STRING_LAU, such as configuration names in PGN 126998, waypoint names in PGN 130074, AtoN names in PGN 129041 where the first byte is a count byte (including the encoding byte), the second byte an encoding byte (differentiating ASCII from Unicode) and subsequent bytes the string itself, I should add that I've never observed any of these types of strings to be null terminated, nor any to be encoded as anything other than ASCII. (One could experiment and see if any commercial devices accept Unicode strings?) 

Decoded samples:

A3 99 2D 80 00 06 53 41 
4C 4F 4F 4E 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 45 (Zone Name)
Unknown: 128
Zone: 0
Zone Name Length: 6
Zone Name: SALOON

A3 99 2D 80 01 07 43 4F 
43 4B 50 49 54 00

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 45 (Zone Name)
Unknown: 128
Zone: 1
Zone Name Length: 7
Zone Name: COCKPIT

A3 99 2D 80 02 06 5A 6F 
6E 65 20 33 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 45 (Zone Name)
Unknown: 128
Zone: 2
Zone Name Length: 6
Zone Name: Zone 3

A3 99 02 80 00 03 00 05 
02 41 4D 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 0
Current Source Id: 3
Source Length: 2
Source Name: AM

A3 99 02 80 01 03 01 05 
02 46 4D 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 1
Current Source Id: 3
Source Length: 2
Source Name: FM

A3 99 02 80 02 03 03 00 
03 53 58 4D 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 2
Current Source Id: 3
Source Length: 3
Source Name: SXM

A3 99 02 80 03 03 02 05 
03 41 75 78 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 3
Current Source Id: 3
Source Length: 3
Source Name: Aux

A3 99 02 80 04 03 02 05 
04 41 75 78 32 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 4
Current Source Id: 3
Source Length: 4
Source Name: Aux2

A3 99 02 80 05 03 05 04 
03 55 53 42 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 5
Current Source Id: 3
Source Length: 3
Source Name: USB

A3 99 02 80 06 03 04 04 
04 69 50 6F 64 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 6
Current Source Id: 3
Source Length: 4
Source Name: iPod

A3 99 02 80 07 03 09 04 
03 4D 54 50 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 7
Current Source Id: 3
Source Length: 3
Source Name: MTP

A3 99 02 80 08 03 0A 25 
02 42 54 00 
Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 8
Current Source Id: 3
Source Length: 2
Source Name: BT

A3 99 02 80 09 03 0E 00 
03 44 41 42 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 2 (Source Name)
Message A: 128
Source Id: 9
Current Source Id: 3
Source Length: 3
Source Name: DAB

A3 99 11 80 07 00 00 00 
00 49 05 05 53 6F 6E 67 
73 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 17 (Folder Name)
Source: 7
Folder/Track Number No: 0
Virtual Folder
Session Id: 5
Folder/Track Name: Songs

A3 99 11 80 07 01 00 00 
00 49 05 06 41 6C 62 75 
6D 73 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 17
Source: 7
Folder/Track Number No: 1
Virtual Folder
Session Id: 5
Folder/Track Name: Albums

A3 99 11 80 07 02 00 00 
00 17 03 43 6C 6F 6F 6B 
61 2D 74 68 65 2D 77 61 
6C 69 65 6E 73 2D 74 72 
69 70 6C 65 6A 75 6E 65 
61 72 74 68 65 64 2D 31 
39 32 6B 2D 31 30 33 39 
36 36 38 31 2D 38 36 39 
34 35 31 36 2D 36 35 30 
34 31 36 2E 6D 70 33 00 

Manufacturer ID: 419 (Fusion)
Industry Code: 4
Message Id: 17
Source: 7
Folder/Track Number No: 2
Music Track
Session Id: 3
Folder/Track Name: looka-the-waliens-triplejunearthed-192k-10396681-8694516-650416.mp3

Elmue, 2025-04-25:
That is interesting.

Kees writes in the XML file that he has no logfiles:

> Unfortunately, these are only used in SonicHub and Fusion brand specific PGNs.
> SonicHub is no longer produced, and for neither do we have any logfiles.
> Therefore the exact way to parse these fields is unclear at the moment.

So yo have logfiles that contain StringLZ ?

Can you please upload a logfile ?
I would like to study the raw data and feed it into my code to test it.

tomer-w, 2025-04-25:
Already did: https://github.com/canboat/canboat/blob/master/samples/fusion-130820.log

This how I tested all the fixes I submitted for the Fusion 130820 PGN.

Elmue, 2025-04-27:
Thank you.
I analyzed your logfile.

Kees writes in the XML comment:

> The length byte includes the zero byte but not itself.

This does not apply to your logfile.
I verified all strings and the length byte contains always the exact length of characters, not including the length byte and not including the zero termination.

> These fields have a length containing what we believe to be the maximum allowed length.

Also this is not correct.
The length byte contains the exact count of characters.
________________________________

Is there a possibility that you check the encoding ?
Can you play for example a Russian band that uses cyrillic characters or a Greek band using Greek characters ?
Then we could see if they use UTF-8 encoding.

tomer-w, 2025-04-27:
Are you use you are fully synced: I'm parsing the sample file and get:
Fusion: Artist Name:  Manufacturer Code = Fusion Electronics; Industry Code = Marine Industry; Message ID = Artist Name; Source ID = 11; Index = 6; Artist = Radiolab

tomer-w, 2025-04-27:
I dont have access to my boat currently. this specific file was supplied by @sbender9. He might be able to produce more samples.

keesverruijt, 2025-04-29:
OK so I'm going to document that STRING_LZ is a Fusion specific string format containing UTF-8, length byte + data + zero terminator with zero terminator not counted in length byte.

tomer-w, 2025-04-29:
Looks good.

sbender9, 2025-04-30:
This is correct. It's actually coming from a SiriusXM channel. Probably a talk channel, Sirius can put pretty arbitrary stuff in there.

sbender9, 2025-04-30:
Also can look strange when coming through BT and listening to a Podcast or something like that.

tomer-w, 2025-04-30:
Agree with @sbender9 . Here is the code which parse this message:
```
public final k0 decodeFromByteStream(byte[] bArr) {
            k0 k0Var = new k0();
            StatusOpEnum decodeFromByteStream = StatusOpEnum.decodeFromByteStream(bArr);
            ve.t.checkNotNullExpressionValue(decodeFromByteStream, "decodeFromByteStream(...)");
            k0Var.setType(decodeFromByteStream);
            k0Var.setSourceID(TCP_receiver.getUInt8(bArr, 2));
            k0Var.setIndex(TCP_receiver.getUInt32(bArr, 3));
            k0Var.setName(TCP_receiver.getString(bArr, 7));
            return k0Var;
```
And in this case the first 2 bytes (the ```setType```) are ```MEDIA_ALBUM_NAME = 32775```

I think this can get closed after the documentation change @keesverruijt already submitted.

Elmue, 2025-04-30:
OK.

Finally I can confirm that they use UTF8.
There is a message
2017-04-15T21:04:07.004Z,7,130820,10,255,31,a3,99,07,80,0b,00,00,00,00,14,4e,50,52,20,e2,80,a2,20,41,70,72,20,31,34,2c,20,32,30,31,37,00

which looks wrong when decoded as ASCII:

[image]

commit 4821fefe (2025-04-12): Fix #498
closed 2025-05-01 by keesverruijt