canboat/canboat #500: Errors in CanBoat.xml

Elmue, 2025-04-07:
I will add errors here as I find them:

1.)
PGN 129540 "GNSS Sats in View",  Parameter "Range residuals"
Unit is missing and Resolution =1 and Min/Max values are wrong.

NMEA document says:
Resolution: 1E-5
Unit: meter
Min = -2.146E4
Max = 2.146E4
___________________________________

keesverruijt, 2025-04-07:
Please use individual issues for all things that you find. Otherwise it is harder for me to react and to keep track of fixes.

keesverruijt, 2025-04-07:
Do you have a reference to the document you refer to?

In this case I think I can use the GPS documentation we use elsewhere.

closed 2025-04-07 by keesverruijt

Elmue, 2025-04-07:
I don't know what you mean with "GPS documentation".

I was googling and found a PDF document named "NMEA 2000 Appendix B.1 -- PGN Table"
It is from 2009, so not so up to date, but very useful.
https://github.com/z-xr/NMEA2000/blob/main/NMEA2000_v1-300_App_B1_PGN_Table%281%29.pdf
It is password protected and texts cannot be copied, but searching works.
Github says "Invalid PDF" which is not true.
You must download it. It cannot be viewed online.

[image]

Elmue, 2025-04-11:
>  it pertains to v1.300 of NMEA 2000

May be. But do you think that this makes the content wrong?
I worked a long time with J1939 where also lots of different versions exist.
Nearly every year new versions is released.

But I have never seen that a new version changes the meaning of parameters in a CAN packet.
The only thing that newer versions do is to 
1.) add new messages
2.) add new fields to the end of existing messages
3.) give "Reserved" or "Spare" fields a new meaning
4.) add new lookup values to the existing ones

But this does not make an old document invalid.
Obviously new information may be missing, but if it says that the "Range residuals" in "GNSS Sats in View" are measured in meters and that the resolution is 1E-5, then this will surely not be changed in any future version by the NMEA because it would break the communication with all CAN controllers that are already in use.

So if this document is illegally uploaded or not should not matter if you are interested in the correctness of your data.
You could also buy this document legally and you will have the same information.

And how can anybody ever proof if you information really comes from reverse engineering or not ?

acolomb, 2025-04-12:
> And how can anybody ever proof if you information really comes from reverse engineering or not ?

Isn't that obvious? Your discussion here with the references to the document and you mentioning how it's okay to use the info whose license is explicitly restricting public use. This discussion is public, like everything about the canboat development repository. Should it ever come to a licensing argument with the NMEA, such mentions of protected material being used during reverse engineering make our arguments futile. It needs to be solely clean-room reverse engineered to legally publish this PGN database.

Has anybody ever tried to contact NMEA about this? Or do we just want to let sleeping dogs lie?

keesverruijt, 2025-04-12:
@acolomb > Has anybody ever tried to contact NMEA about this? 

I have, in 2009-2012, repeatedly spoken with NMEA technical staff (at the time, one person.) We openly discussed me buying the standard and how this would limit me publishing information. Answer: yes, the NMEA standard including the PGN appendices A and B is licensed in such a way that you cannot then openly publish (all) its content.

So the NMEA is (or at least was) quite aware of my project (that I had already started and published at that time) and they grudgingly conceded that reverse engineering was not impossible and that as long as the project steers commercial developers towards buying the standard, that was okay with them.

In fact, several manufacturers have worked closely with me in the past and the understanding has always been the same: I will not get access to the standard. I have asked a few pointed questions here and there, and it was up to them to answer. They have sometimes told me that they could not answer, and sometimes they could.

@Elmue > You could also buy this document legally and you will have the same information.

No, I cannot legally and thus morally do so. The Appendices A and B (which is what manufacturers tell me they contain) are licensed such that all people with knowledge of them are bound by the license that, apparently, contains limitations to the use of the standard.

@Elmue > And how can anybody ever proof if you information really comes from reverse engineering or not ?

That is trying to prove the improvable. I can say for myself that I have not, but the same may not be true for one or more of my contributors. I have never entertained pull requests that fix so many items as to it being suspicious, but that would be on them. You'll have to do with my statement and my moral compass.

At the time I also spoke with manufacturers that were privately also not happy with all this secretive stuff, but they apparently never convinced the NMEA of the errors of their ways. 

Unfortunately the closed format has weakened the standard in multiple ways. Lack of peer review means that there are many parts where the PGNs are more complicated than needed; multiple string encodings come to mind but the worst part are the DSC PGNs. See https://web.archive.org/web/20160408121445/http://www.nmea.org/Assets/20150328%20dsc%20technical%20corrigendum%20v2.database%20version%202.100.pdf (I will update the link in CANboat as well, it is no longer available at the original URL.)

From that document, and I quote:

> This Technical Corrigendum refers to PGN 129808. It was recently discovered that PGN
> 129808 in NMEA Network Message Database Version 2.100 February 2015 (NMEA
> 2000 Appendix B), has erroneous fields, data dictionary items and field names.
> THIS PGN 129808 IN THE NMEA NETWORK MESSAGE DATABASE
> VERSION 2.100 FEBRUARY 2015 SHOULD NOT BE USED
> Manufacturers should revert to NMEA Network Message Database Version 2.000 and
> in conjunction use the NMEA Corrigendum TC 2000 20130720.

and then it goes on how to interpret the DSC data, which is quite convoluted.

Elmue, 2025-04-13:
I find it weird that they are so closed with their documents but on the other hand there are these documents in the wayback machine as the one you link here about PGN 129808 which was published in 2016 and shows all details of the message.
I would understand that these documents are hidden behind a login for users who have paid the license.
But they published them freely so wayback machine could copy them.
How is this to explain?

Elmue, 2025-04-13:
Thanks to this document I finally understand how to decode the MMSI.
I did it as explained in your XML document, but your explanation is wrong.
Therefore I got invalid hex characters which were not decimal digits.

This is not BCD encoding. 
They use a very strange encoding of a decimal number that I have never seen before.
BCD encodes each digit separately, but NMEA encodes byte-wise.

You write:
"... so "1234" is represented by 2 bytes containing 0x12 and 0x34."
(This in deed would be BCD encoding)

But the NMEA document says: "0x0C ( = 12 ), 0x22 ( = 34 )"
... so "1234" is represented by 2 bytes containing 0x0C and 0x22 !

Now I can decode correctly your logfile pgn129808.raw and get the MMSI number 512000953

And you write:
"A number with an odd number of digits will have 0 as the first digit in the first byte."

Also this is not correct.
The zero is added to the end.

NMEA says: 
"Each two decimal digits shall be encoded as the equivalent value in hexadecimal. As an MMSI number is nine (9) digits long a **trailing** zero will be added."

commit 9aebce61 (2025-04-07): Fix #500