canboat/canboat #502: Introduce ISO_NAME fieldtype

keesverruijt, 2025-04-07:
See #496 

PGN fields should be recognisable as having a distinct content, so that explaining software can use a special lookup type to show the meaning of the value. Adding a PGN field type is useful therefore.

keesverruijt, 2025-04-13:
Should be fixed in commit 6f6942d955556c74efb482351dac68a56d982f87

@Elmue, @sbender9  -- see changes in generated XML and JSON. Also, please tell me what you think about how analyzer now handles this. There is a (fake) ALERT message in `analyzer/tests/pgn-126983.raw`, throw this at analyzer with `-json -nv` to see what happens now (or look at the corresponding `.out` file.) (Plain analyzer also does something similar, but that has no parsing consequences.)

sbender9, 2025-04-13:
looks like pgn-126983.in was not checked in?

 ../../rel/darwin-arm64/analyzer -json -nv < pgn-126983.in > /tmp/pgn-126983-nv.out -fixtime pgn-test 2> /tmp/pgn-126983-nv.err
/bin/sh: pgn-126983.in: No such file or directory

keesverruijt, 2025-04-13:
Sorry, they're in now.

keesverruijt, 2025-04-13:
"Maretron" is now back to just "Maretron", for instance.

sbender9, 2025-04-13:
ISO_NAME stuff looks good. No effect on canboatjs right now, but I need to implement this too at some point...

Elmue, 2025-04-16:
Excellent.
You also added more manufacturers and even new PGNs.

You did nearly all the changes for which I opened an issue.
Now I can remove some workarounds from my code.

But issues is still remaining:
In both 129808 entries there should be
```
<FieldType>LOOKUP</FieldType>
<LookupEnumeration>DSC_CATEGORY</LookupEnumeration>

```
for field DSC Category.
___________________________________

Another issue remaining:
In PGN 130060 the second field "PGN" is still declared as type NUMBER instead of PGN.
___________________________________

In PGN 130061 there is a field "Source NAME" which is still declared as NUMBER.
___________________________________

Another thing which is not 100% correct is your maximum value for the new PGN fields.
You defined a maximum value of 131071 which is 0x1FFFF
But PGNs can reach up to 0x3FFFF  (which is 262144)
The highest bit is still reserved but some day they may extend to this range.

The ID of the 29 bit CAN packet in J1939 is defined as:
3 bit priority + 18 bit PGN + 8 bit source address = 29 bit
and 2 ^ 18 is 262144 

[image]

___________________________________

You have several explanations of your fields in the XML which is very good.
But there are fields that I don't understand because there is no explanation.

What is the purpose of your new field  "PartOfPrimaryKey"  ?
What is the "primary key" ?
And how are the "parts" to be set together to build a key?
Some of your "PartOfPrimaryKey" have also a Match and first I thought it has to do with matching.
But then I found even strings declared as "PartOfPrimaryKey" !
This new field is really an enigma to me.

Also I don't understand why you have additionally to "BitOffset" a value "BitStart"  ?
What is this value needed for ?
It would be good to explain this.

_____________________________________

Another really tiny improvment would be to add to PGN 60928, last field "Arbitrary address capable":
```
<FieldType>LOOKUP</FieldType>
<LookupEnumeration>YES_NO</LookupEnumeration>

```
____________________________________

> Also, please tell me what you think about how analyzer now handles this.

I never compiled this code.
I use my own software that I'am developing to load and parse your XML file.

Elmue, 2025-04-18:
It says:

> Some fields are not byte-aligned, and will contain a non-zero value
> for BitStart, indicating at which bit in the byte the field starts.
> Note that fields can span multiple bytes. In practice we have not
> seen fields that start at a non-zero BitStart and span more than 2 bytes.

But the information in BitOffset is fully sufficient.

So you are saying that 
BitStart = BitOffset AND 7 ?
and
ByteStart = BitOffset / 8

If I have a BitOffset = 40 it means that the value must be extracted starting at bit 0 in byte 5 because 40 &amp; 7 = 0
If I have a BitOffset = 41 it means that the value must be extracted starting at bit 1 in byte 5 because 41 &amp; 7 = 1
If I have a BitOffset = 47 it means that the value must be extracted starting at bit 7 in byte 5 because 47 &amp; 7 = 7
If I have a BitOffset = 48 it means that the value must be extracted starting at bit 0 in byte 6 because 48 &amp; 7 = 0
If I have a BitOffset = 49 it means that the value must be extracted starting at bit 1 in byte 6 because 49 &amp; 7 = 1

I still don't understand why you added this additional field, if it is so simple to calculate?

Everybody who uses your XML will need a routine anyway that extracts for example 11 bits Manufacturer code out of a stream of bytes. The only needed information is BitOffset and BitLength to do that.

keesverruijt, 2025-04-22:
I don't remember, but the original XML was not my own concoction. It started about 15 years ago when I was approached by an Australian professor at a university that was writing a project called OpenSkipper, and he wanted to use my definitions which were in C only at the time. A lot of the superfluous fields came from their requests.

Elmue, 2025-04-22:
I already supposed that when I downloaded OpenSkipper from Github.
Your old file pgns.xml seems to come from OpenSkipper.
This XML file was FULL of errors and missing data.
Many parameters beeing total crap.
I thought I will need months to fix all these errors.

But then luckily I found CanBoat where you already did all this tremendous work.
You created a completely new XML format where lookups are separated from parameters which is much more intelligent than the old XML format where each lookup was repeated over and over again.

keesverruijt, 2025-04-22:
> Your old file pgns.xml seems to come from OpenSkipper.

No, he asked politely if I could create an XML format in addition to the text format that the code produced at the time. The original format, as you say, was very repetitive.