canboat/canboat #504: Handling of Lookup values

Elmue, 2025-04-09:
In the XMl file you write:

```
<FieldType Name="NUMBER">
   <EncodingDescription>
Number fields that use **two or three bits** use one special encoding, for the maximum value.  
When present, this means that the field is not present.

Number fields that use **four bits** or more use two special encodings.
The maximum positive value means that the field is **not present**. 
The maximum positive value minus 1 means that the field has an **error**. 
For **signed numbers** the maximum values are the maximum positive value and that minus 1, not the all-ones bit encoding which is the maximum negative value.

```
Can you cite any official NMEA document where these details are explained ?
Or is this your own observation ?

I observe the **exact oppposite** of what you write about signed numbers:
Serveral controllers in your logfiles send signed parameters where FFFF means "Not Available".
If I would decode this as a signed value the display would be -1 which does not make sense.
May be you are right and the rule is as you write, but many controllers do not comply with this rule.
They send the maximum UNSIGNED value to indicate "Not Available".

For example for PGN 129540 "GNSS Sats in View" I see the packet:
19FA041C: 1B FF 0F 01 85 07 9C 18 04 10 FF FF FF FF FF  ...... and more bytes .....
The last 5 bytes are FF. They correspond to:

		<Field>
		 <Name>Range residuals</Name>
		 <BitLength>32</BitLength>
		 <Resolution>1E-5</Resolution>
		 <Unit>m</Unit>
		 <Signed>true</Signed>
		 <FieldType>NUMBER</FieldType>
		</Field>
		<Field>
		 <Name>Status</Name>
		 <BitLength>4</BitLength>
		 <Resolution>1</Resolution>
		 <Signed>false</Signed>
		 <FieldType>LOOKUP</FieldType>
		 <LookupEnumeration>SATELLITE_STATUS</LookupEnumeration>
		</Field>
		<Field>
		 <Name>Reserved</Name>
		 <BitLength>4</BitLength>
		 <FieldType>RESERVED</FieldType>
		</Field>

It would not make much sense to display "Range residuals" as a permanent (never changing) value of -0.00001 m
So I would say it is the opposite of what you write: The maximum UNSIGNED value means "Not Available" no matter if the value is signed or unsigned.
___________________________________________________

And I suppose that this rule must also apply to Lookup values.
I decode your logfile dirona-actisense-serial.raw
and there is a packet 0DF20DD0: 0A 40 00 20 45 00 00 00
which decodes to this:

[image]

As you see there is a value 2 for Indicator 11 which is not defined in

```
<LookupEnumeration Name='OFF_ON' MaxValue='3'>
  <EnumPair Value='0' Name='Off' />
  <EnumPair Value='1' Name='On' />
</LookupEnumeration>

```
So my question is if there is any official documentation.

I suppose that the above rule for NUMBER also applies to LOOKUP 
and not only if 4 bits are present, but even if only **2 bits** are present!

The next  question is where does the logfile dirona-actisense-serial.raw come from ?
Is it possible that the CAN controller has reported the value 2 as a temporary Error here for Indicator 11 ?
Later in the logfile this value of 2 disappears.
Can you give me some explanation on this ?

keesverruijt, 2025-04-10:
> So my question is if there is any official documentation.

For what? You do know that the NMEA 2000 standard is closed and unavailable to me and all others that want to write open source, right?

> I suppose that the above rule for NUMBER also applies to LOOKUP
> and not only if 4 bits are present, but even if only 2 bits are present!

For some fields, yes. Maybe this rule needs adapting.

> The next question is where does the logfile dirona-actisense-serial.raw come from ?

As it says, from a MV called Dirona :-) whose owner, in this case, I happen to know.

> Is it possible that the CAN controller has reported the value 2 as a temporary Error here for Indicator 11 ?

No idea, quite likely.

> Later in the logfile this value of 2 disappears.
> Can you give me some explanation on this ?

No.

keesverruijt, 2025-04-10:
> There are enum values where the rule of the maximum value does not apply:
> 
> 	<LookupEnumeration Name='TIME_STAMP' MaxValue='63'>
> 	  <EnumPair Value='60' Name='Not available' />
> 	  <EnumPair Value='61' Name='Manual input mode' />
> 	  <EnumPair Value='62' Name='Dead reckoning mode' />
> 	  <EnumPair Value='63' Name='Positioning system is inoperative' />
> 	</LookupEnumeration>
> 
> Here the Maximum value 63 does not mean "Not Available".
> And the Maximum -1 value 62 does not mean "Error".
> But this is no problem.

Nope. NMEA 2000 is not a math system that always works the same, apparently. At least, that is my guess.

If you think about it, the maximum value does mean the same, in a sense. You could argue that 0x3f in this field means something for the other fields, but I would surmise that in that case the other fields will also contain their own "Invalid" value.

keesverruijt, 2025-04-10:
On error and unavailable values:

> Can you cite any official NMEA document where these details are explained ?

No, of course I can't. They keep the entire system under wraps, to keep outsiders out.

> Or is this your own observation ?

Yes. But the description is not 100% and could be improved to add a little more uncertainty as it is not a 100% rule. It should say "in general", and that for 2 bit fields it may vary. 

The documentation should also explain that lookup values are sometimes just guesses, and that sometimes they are derived from proper sources, and that this is not documented per lookup type (too much work.)

Some lookup types override the special meanings, or may do so. We don't really know. All we can do is observe a lot, and then cook up a distillation of what we see.

keesverruijt, 2025-04-10:
On max values for unsigned/signed:

> I observe the exact oppposite of what you write about signed numbers:
> Serveral controllers in your logfiles send signed parameters where FFFF means "Not Available".
> If I would decode this as a [...]

> ... goes on about newly changed field def of range residuals, because of yourself, that shows up as FFFF.

This actually hints that field is NOT signed, but unsigned.

In fact, this is how we catch a lot of fields that have been guesstimated wrongly: a signed field will show up like 0x7F or 0x7FFF etc. 

Here are some extracts of the `samples/` that show my POV:

```
$ ./rel/darwin-arm64/analyzer -data -q < samples/pgn129538.raw 
2024-12-25-00:00:01.000 4   8 255 129538 GNSS Control Status:  69 03 FF 7F FF 7F FF 7F 0B FF FF 40 FC FF FF FD
2024-12-25-00:00:01.000 4   8 255 129538 GNSS Control Status:   i  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
2024-12-25-00:00:01.000 4   8 255 129538 GNSS Control Status:  SV Elevation Mask = 5.0 deg; PDOP Mask = Unknown; PDOP Switch = Unknown; SNR Mask = Unknown; GNSS Mode (desired) = Auto; DGNSS Mode (desired) = SBAS if available; Position/Velocity Filter = 0; Max Correction Age = Unknown; Antenna Altitude for 2D Mode = -9.60 m; Use Antenna Altitude for 2D Mode = Unknown
```

and even more telling, one of the most used PGNs, 128267 "Water Depth" from depth sounders, which often doesn't contain the sensor depth offset:

```
./rel/darwin-arm64/analyzer -data -q < samples/susteranna2020.raw 128267 | tail -n 16
ERROR 2025-04-10T08:06:37.544Z [analyzer] Error reading message, scanned 18446744067546404960 bytes from 2020-08-22T13:52:36.981Z,0,16712965,255,0,0
ERROR 2025-04-10T08:06:37.545Z [analyzer] Unknown message error 2: '2020-08-22T13:52:36.981Z,0,16712965,255,0,0
'
ERROR 2025-04-10T08:06:37.546Z [analyzer] Error reading message, scanned 18446744067546404960 bytes from 2020-08-22T13:52:43.993Z,7,65301,255,0,0
ERROR 2025-04-10T08:06:37.546Z [analyzer] Unknown message error 2: '2020-08-22T13:52:43.993Z,7,65301,255,0,0
'
2020-08-22T13:52:57.995Z 3  35 255 128267 Water Depth:  SID = Unknown; Depth = Unknown; Offset = 0.590 m; Range = 120 m
2020-08-22T13:52:59.157Z 3  17 255 128267 Water Depth:  FF FF FF FF FF FF 7F FF
2020-08-22T13:52:59.157Z 3  17 255 128267 Water Depth:   .  .  .  .  .  .  .  .
2020-08-22T13:52:59.157Z 3  17 255 128267 Water Depth:  SID = Unknown; Depth = Unknown; Offset = Unknown; Range = Unknown
2020-08-22T13:52:59.816Z 3  15 255 128267 Water Depth:  00 FF FF FF FF 00 00 FF
2020-08-22T13:52:59.816Z 3  15 255 128267 Water Depth:   .  .  .  .  .  .  .  .
2020-08-22T13:52:59.816Z 3  15 255 128267 Water Depth:  SID = 0; Depth = Unknown; Offset = 0.000 m; Range = Unknown
2020-08-22T13:53:01.004Z 3  35 255 128267 Water Depth:  FF FF FF FF FF 4E 02 0C
2020-08-22T13:53:01.004Z 3  35 255 128267 Water Depth:   .  .  .  .  .  N  .  .
2020-08-22T13:53:01.004Z 3  35 255 128267 Water Depth:  SID = Unknown; Depth = Unknown; Offset = 0.590 m; Range = 120 m
2020-08-22T13:53:02.041Z 3  35 255 128267 Water Depth:  FF FF FF FF FF 4E 02 0C
2020-08-22T13:53:02.041Z 3  35 255 128267 Water Depth:   .  .  .  .  .  N  .  .
2020-08-22T13:53:02.041Z 3  35 255 128267 Water Depth:  SID = Unknown; Depth = Unknown; Offset = 0.590 m; Range = 120 m
2020-08-22T13:53:02.193Z 3  17 255 128267 Water Depth:  FF FF FF FF FF FF 7F FF
2020-08-22T13:53:02.193Z 3  17 255 128267 Water Depth:   .  .  .  .  .  .  .  .
2020-08-22T13:53:02.193Z 3  17 255 128267 Water Depth:  SID = Unknown; Depth = Unknown; Offset = Unknown; Range = Unknown
```

In particular, look at that last sentence: it shows an U32 depth, e.g. 0xFFFFFFFF, and an I16 offset 0x7FFF.

In other words, IMO your "signed value max value" analysis is incorrect.

keesverruijt, 2025-04-10:
As to range residuals showing up as 0xFFFFFFFF from some sensors, my belief is that those sensors are incorrect.

If you look at my samples directory, you will find many sources for this PGN, some with 0x7FFFFFFF and one with 0xFFFFFFFF:

```
$ grep -l '129540.*ff,ff,ff,7f,' samples/*.raw
samples/dirona-actisense-serial.raw
samples/merrimac-2022.raw
samples/merrimac-actisense-serial-2011.raw
samples/scx20.raw
samples/susteranna-actisense-serial.raw
$ grep -l '129540.*ff,ff,ff,ff,' samples/*.raw
samples/dirona-actisense-serial.raw
```

If we zoom in on the dirona file, we find that it has two GPS sources, source ID 7 that uses (incorrect) unsigned value and ID 33 that uses a correct value.

In this case, I place the highest trust in the `scx20.raw` file, which is of my own Furuno SCX-20, which is a recent device that is certified by the NMEA. It should be compliant with v2.000, but I can't prove that right now.

The takeaway from all of this: manufacturers make mistakes, and log files from the internet where you do not know who manufactured a device (to know if it is a _recently certified_ device) should not be trusted.

Elmue, 2025-04-11:
> You do know that the NMEA 2000 standard is closed and unavailable to me and all others that want to write open source, right?
>  They keep the entire system under wraps, to keep outsiders out.

It is not unavailable. It is only expensive:
https://www.csselectronics.com/products/nmea-2000-dbc-file-pgn-database

There are also programmers who receive a good salary and in their spare time they work on their open source projects as a hobby. Not everybody who writes open source is automatically also poor.
Many Linux developers have a regular job where they earn their living and in their free time they develop Linux.
Therefore my assumption that you might possess the official NMEA message declarations is not so absurd.

>>  They send the maximum UNSIGNED value to indicate "Not Available".
>This actually hints that field is NOT signed, but unsigned.

There are 3 options to interpret the data of FFFF that we observe on CAN bus:
1.) This field is unsigned and FFFF means "Not Available" (so the `<Signed>true</Signed>` is wrong)
2.) This field is in deed signed and the rule is wrong (Signed values are allowed to send Unsigned FFFF for "Not Available")
3.) The CAN bus controller does not comply with the rules

I would assume that this field is definitely signed because:
1.) the NMEA document says so.
2.) on this page you see examples for Range residuals which are also negative:
-13.8, -1.9, 11.4, -33.6, 0.9, 6.9, -12.6, 0.3, 0.6, -22.3
https://docs.novatel.com/OEM7/Content/Logs/GPGRS.htm

So I think that we must consider to accept the maximum unsigned value as "Not available" for signed signals.
And it makes even sense because is simplifies the rule:
Maximum raw value for signed or unsigned values means "Not Available"
This allows to discard "Not Available" fields even before you do the calculation with Resolution and Offset by just looking at the raw value.

But this does not say that the value in `<RangeMax>2.147E4</RangeMax>` should be ignored.
If the signed value exceeds this limit, the parameter is also "Not Available".

keesverruijt, 2025-04-12:
Again, I point to the SCX20 logfile. This is a recent (>= 2020) GNSS compass from Furuno that has been certified.

And what do we see here: proper `FF FF FF 7F` bytes for range residuals, corroborating my thesis:

```
2021-01-27T23:34:19.845Z 6   1 255 129540 GNSS Sats in View:  FF FF 0E 01 08 2C 26 20 88 13 FF FF FF 7F F5 03 A2 10 40 5E CC 10 FF FF FF 7F F5 07 17 35 23 A0 C0 12 FF FF FF 7F F5 08 F0 0A FB 19 A0 0F FF FF FF 7F F5 09 D3 0B AF 86 04 10 FF FF FF 7F F5 0E D1 0F 00 E1 CC 10 FF FF FF 7F F5 11 00 13 22 BC 5C 12 FF FF FF 7F F5 13 85 08 B7 AE 3C 0F FF FF FF 7F F5 15 D2 19 89 19 30 11 FF FF FF 7F F5 16 C7 12 B3 4A 94 11 FF FF FF 7F F5 1C 5A 0C A2 DD CC 10 FF FF FF 7F F5 1E E6 22 1D D3 94 11 FF FF FF 7F F5 2C 2E 17 E5 AF 94 11 FF FF FF 7F F5 2E 00 0F A0 B2 A0 0F FF FF FF 7F F5
2021-01-27T23:34:19.845Z 6   1 255 129540 GNSS Sats in View:   .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  5  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  0  .  .  .  .  .  .  .  .  .  .  J  .  .  .  .  .  .  .  .  Z  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
2021-01-27T23:34:19.845Z 6   1 255 129540 GNSS Sats in View:  SID = Unknown; Range Residual Mode = Unknown; Sats in View = 14; PRN 1 = 1; Elevation 1 = 64.6 deg; Azimuth 1 = 47.2 deg; SNR 1 = 50.00 dB; Range residuals 1 = Unknown; Status 1 = Used+Diff; PRN 2 = 3; Elevation 2 = 24.4 deg; Azimuth 2 = 138.2 deg; SNR 2 = 43.00 dB; Range residuals 2 = Unknown; Status 2 = Used+Diff; PRN 3 = 7; Elevation 3 = 77.9 deg; Azimuth 3 = 234.9 deg; SNR 3 = 48.00 dB; Range residuals 3 = Unknown; Status 3 = Used+Diff; PRN 4 = 8; Elevation 4 = 16.0 deg; Azimuth 4 = 38.1 deg; SNR 4 = 40.00 dB; Range residuals 4 = Unknown; Status 4 = Used+Diff; PRN 5 = 9; Elevation 5 = 17.3 deg; Azimuth 5 = 197.6 deg; SNR 5 = 41.00 dB; Range residuals 5 = Unknown; Status 5 = Used+Diff; PRN 6 = 14; Elevation 6 = 23.2 deg; Azimuth 6 = 330.0 deg; SNR 6 = 43.00 dB; Range residuals 6 = Unknown; Status 6 = Used+Diff; PRN 7 = 17; Elevation 7 = 27.9 deg; Azimuth 7 = 275.9 deg; SNR 7 = 47.00 dB; Range residuals 7 = Unknown; Status 7 = Used+Diff; PRN 8 = 19; Elevation 8 = 12.5 deg; Azimuth 8 = 256.3 deg; SNR 8 = 39.00 dB; Range residuals 8 = Unknown; Status 8 = Used+Diff; PRN 9 = 21; Elevation 9 = 37.9 deg; Azimuth 9 = 37.5 deg; SNR 9 = 44.00 dB; Range residuals 9 = Unknown; Status 9 = Used+Diff; PRN 10 = 22; Elevation 10 = 27.5 deg; Azimuth 10 = 109.6 deg; SNR 10 = 45.00 dB; Range residuals 10 = Unknown; Status 10 = Used+Diff; PRN 11 = 28; Elevation 11 = 18.1 deg; Azimuth 11 = 325.1 deg; SNR 11 = 43.00 dB; Range residuals 11 = Unknown; Status 11 = Used+Diff; PRN 12 = 30; Elevation 12 = 51.2 deg; Azimuth 12 = 309.7 deg; SNR 12 = 45.00 dB; Range residuals 12 = Unknown; Status 12 = Used+Diff; PRN 13 = 44; Elevation 13 = 34.0 deg; Azimuth 13 = 258.0 deg; SNR 13 = 45.00 dB; Range residuals 13 = Unknown; Status 13 = Used+Diff; PRN 14 = 46; Elevation 14 = 22.0 deg; Azimuth 14 = 262.0 deg; SNR 14 = 40.00 dB; Range residuals 14 = Unknown; Status 14 = Used+Diff
```

I do agree with you that even 2 bit fields actually have two special meanings, normally. I will adapt this accordingly.

keesverruijt, 2025-04-13:
> > This is a recent (>= 2020) GNSS compass from Furuno that has been certified.
> 
> This has nothing to do with beeing certified or not. It also has nothing to do with the NMEA version for which it was certified or the year when it was certified.
> 
> They cannot use the value FFFF to indicate that a signed parameter is not available because this value has already the meaning of -1. This is definitely a bug. And it has always been a bug no matter what NMEA version. A raw value of FFFF for a signed 16 bit parameter means always -1 and cannot mean that the parameter is not available. They must use 7FFF or anything outside the Min/Max range to indicate this. If they do not there is a programming error in the firmware.

My point is that the Furuno SCX-20 does it properly, they use FF FF FF 7F for a not-present (unknown) range residuals. And that this shows that signed values are using the maximum unsigned value for unknown values, not all-ones.

> 
> By the way: In your Dirona logfile there are several controllers sending the battery temperature in PGN 127508.
> 
> Controller B0 from Maretron sends correctly 33 degree Celsius But controllers 04 and 05 from Northern Lights send 0x0000 (zero Kelvin) instead of 0xFFFF for not available. The worst is controller 3C from Mastervolt that sends a battery temperature of 128 degree Celsius ! This battery would already be melting away or be on fire.
> 
> What an ugly buggy crap! Worst quality of the firmware.

The entire PGN has been obsoleted I think, and remember that this file is from over 10 years ago.

You should see early Raymarine devices spit complete junk!

> I'am absolutely not surprised that CAN bus controllers send invalid data. We must be prepared for crappy data and handle it correctly.
> 
> As a temperature of zero Kelvin is impossible on planet Earth, we must interpret this also as "Not available".

No it is not. I agree it is unlikely that you will use NMEA 2000 for anything in a low temperature lab, but the Dutchman Kamerlingh Onnes already reached < 1 K in the early 1900s.

> 
> I recommend you to change RangeMin to 1 Kelvin for **ALL** temperatures in the XML file because zero Kelvin can only mean "Not Available". Not even 100 Kelvin is possible on planet Earth which would be -170 °Celsius.
> 
> An exception would be theoretically parameters that are signed and show a temperature difference or a temperature change. But such a parameter does not exist in the entire XML file.
> 

No, I won't. There are whole heaps of fields where values are "unlikely", like a tank volume of more than a supertanker can contain, etc. or speeds over 100 kn, etc etc.

My motto here: crap in = crap out.

The "minRange" and "maxRange" are not about what is likely, it is what the minimal/maximal value in the PGN means.

I'd rather propagate the crap than filter it.  But hey, feel free to shoot the user in the head if they dare to analyse such a device with your tool. Whatever takes your fancy.

> Just another tiny correction for which I don't want to open a new issue:
> 
> GN 127506 "DC Detailed Status" --> parameter "State of Charge" and PGN 127491 "Electric Energy Storage Status, Dynamic" --> parameter "State of Charge"
> 
> Your Dirona logfile reports a value of "100". I recommend to add:
> 
> `<Unit>%</Unit>`

Good point.

Elmue, 2025-04-17:
>  but the Dutchman Kamerlingh Onnes already reached < 1 K in the early 1900s.

Haha!  ;-)
And he connected a CAN bus temperature sensor in his experiment ?

> My motto here: crap in = crap out.

I do not agree with this.
Zero Kelvin definitely means "Not Available".
It **CAN** NOT mean anything else.
This is what I tried to explain.
Obviously it is caused by a sloppy programming.
Sloppy programmers do not send the maximum value as they should.

But please put yourself into the user that connects my software to his CAN bus and sees that the battery has a temperature of -273 degree.
The consequence will be that he says: "This software works like shit."
It will not come into his mind that the temperature sensor sends nonsense data.
He will blame it on my software.
"crap in = crap out" means that my software displays crap.
I prefer to display "Not available" than -273 degree.

If you don't want to implement a minimum value of 1 Kelvin for all temperatures and you also don't want to convert Kelvin into Celsius, it is your choice, it is your database.
I already do this in my code. So this is no problem for me.

But I cannot see ANY sense to show a battery temperature of -273 degree Celsius to my users in my high quality software.

sbender9, 2025-04-17:
Definitely agree with @keesverruijt on this one. Zero is a valid value for temperature, we should not be changing our definitions because there are some bad actors out there.

keesverruijt, 2025-04-17:
> I prefer to display "Not available" than -273 degree.

Since your users are people that want to analyse a CAN bus, I think they should deserve the truth, whatever it is. And the truth is that all-zeros is NOT "Not Available". It is "0.0 degrees Kelvin, which is unlikely".

If we don't use 0, then what value do we consider "reasonable". And if we say, for sake of argument, that temperatures outside the range of 243 to 473 K are "likely incorrect", where do we stop? Do we start limiting speed values? Or depths? Pressures? Geographical positions that are > 90 N/S or > 180 E/W?

This minimalistic approach is also why the units are what they are: the source uses these. So it's 95% SI units, the main exception being geographical positions which are in degrees instead of radians, because that is how they appear in the PGN. 

And yes, if you want to show this to a user, you need to convert this. My own tools do this. Yours can as well.

Elmue, 2025-04-18:
@keesverruijt and @sbender9:

You both did not get my point.

> Do we start limiting speed values? Or depths? Pressures? Geographical positions that are > 90 N/S or > 180 E/W?

If you read what I wrote you will notice that I never said that.
I never talked about speeds or depths or pressures or the longitude making sense or not making sense.
Obviously also these values may be wrong.
But that is not the point.
I do not want to check all values if they make sense.

I'am only talking about detecting firmware errors the are **obvious**.
And I'am only talking about the extreme raw values which are the raw minimum, zero and the raw maximum.
For a 16 bit values these would be 0000, FFFF and in case of a signed value also 7FFF.
Nothing more.

What if the developer of a firmware made an error and sends zero instead of maximum values?
In case of speed, depth, pressure or longitude we can NOT detect this because zero is a valid value.

But on a vessel there will never be a temperature of zero Kelvin except you have Dutchman Kamerlingh on board and is making his experiments and connects a CAN bus temperature sensor to his experiment - which will never happen.

I think that nobody can deny that 0 Kelvin is an impossible temperature.
In case of PGN 127508 it means that there is no temperature sensor connected to measure the battery temperature.
Therefore I display "Not available" instead of -273 degree Celsius.

sbender9, 2025-04-18:
I totally get your point, and it makes sense, but as I think Kees was trying to point out, this would open a huge can of worms for canboat. We should stick with what the n2k specification says.

UI implementers should handle special cases like this is. And I am one of those, I have many hundreds of users and this has never come up.

Elmue, 2025-04-21:
> I have many hundreds of users and this has never come up.

What do you mean ?
You have never seen a CAN controller sending a temperature of zero Kelvin ?

As I wrote above: Dirona logfile
Controllers 04 and 05 from Northern Lights send 0x0000 (zero Kelvin) instead of 0xFFFF for not available.

keesverruijt, 2025-04-22:
Yes, those Northern Lights controllers are apparently really bad at following the standard. It has incorrect values for a large number of fields.

closed 2025-04-22 by keesverruijt