canboat/canboat #506: Wrong information on your site

Elmue, 2025-04-09:
[image]

keesverruijt, 2025-04-10:
The proper way to address this is:

- Write a proper description
- Add some proof by showing a CAN log, in a format that is digestible by `analyzer`, that shows your case that analyzer analysis is incorrect.

I'm not sure where you got that screenshot, at the very least it may be out of date for all I know. If it is copyrighted material, I don't want to know.

Elmue, 2025-04-11:
You declare on your site that the PGN 1EE00 is Single Frame.
But this is wrong.
I wrote "Correct : 1EDFF" into the image.

The range must be
1ED00 - 1EDFF = Single Frame
not
1ED00 - 1EE00
which is wrong.

1EE00 is Fast Packet

keesverruijt, 2025-04-12:
Actually, you are still wrong: 1ED00 is also fast packet, so the only thing that needs to be fixed is change 'single frame' to 'fast packet' on that line.

closed 2025-04-12 by keesverruijt

Elmue, 2025-04-13:
You are right !

commit 4d66fc7c (2025-04-12): Fix #506
commit b1642b10 (2025-04-12): Fix #506