canboat/canboat #507: XML file shows nonsesne Manufacturer and Industry for not yet reverse engineered PGN 61184

Elmue, 2025-04-09:
You have in the XML file 8 dummies for unknown messages:

```
<PGNInfo>
  <PGN>61184</PGN>
  <Id>0xef00ManufacturerProprietarySingleFrameAddressed</Id>
  <Description>0xEF00: Manufacturer Proprietary single-frame addressed</Description>
  <Explanation>Manufacturer proprietary PGNs in PDU1 (addressed) single-frame PGN 0xEF00 (61184). 
  When this is shown during analysis it means the PGN is not reverse engineered yet.</Explanation>

```
Some of them contains 3 fields for Manufacturer Code, Industry Code and Data

I found a logfile from a boat in internet that I attach here:

[CAN Trace from Boat.zip](https://github.com/user-attachments/files/19675257/CAN.Trace.from.Boat.zip)

When you run this logfile you see that for ALL messages 61184 a complete garbage is displayed:

[image]

As you see address 1C is an engine controller from E Controls that is from industry group Marine as the NAME field shows.

It also sends PGN 0xEF00 where your XML file decodes complete nonsense from the unknown data.
The Industry is decoded as "Agriculture", but this logfile is clearly from a boat.
And the controller at address 17 also sends this message and here the unknown data is decoded as Industry Group Highway.

And when you run this logfile you see that these values are permanently changing.
With each packet another nonsense is displayed:

[image]

It really changes with each and every packet in the logfile and it is 100% garbage.

So I strongly recommend to remove the fields Manufacturer Code, Industry Group and Data not only from
message 61184 but also from 61440 and 65280 and 126720.

A vendor can use this message to send ANY data.
You cannot assume that the first 11 bits in each packet are the Manufacturer is Code and that bits 13 to 15 are Industry Group.
This is wrong.

If I were you I would remove all these messages completely because they have no useful benefit:

```
0x0E800:  59392 --> only a binary block is displayed (useless)
0x0EF00:  61184 --> Manufacturer, Industry Group are displayed wrongly
0x0F000:  61440 --> Manufacturer, Industry Group are displayed wrongly
0x0FF00:  65280 --> Manufacturer, Industry Group are displayed wrongly
0x1ED00: 126208 --> only a binary block is displayed (useless)
0x1EF00: 126720 --> Manufacturer, Industry Group are displayed wrongly
0x1F000: 126976 --> only a binary block is displayed (useless)
0x1FF00: 130816 --> only a binary block is displayed (useless)

```

keesverruijt, 2025-04-10:
No, a vendor is not supposed to send proprietary messages with incorrect manufacturer fields. If he does, he is in violation of the NMEA 2000 standard.

If you ask me nicely (you seem to be very passive aggressive in your messages, I wonder if you realise that?) I can explain this to you. But if you think about it, you can come up with the reason yourself.

Interesting that you say that they have no benefit to you -- I just honoured a request from somebody using `canboat.json` not to filter them out there, as he was missing them there. They serve as a placeholder for all PGNs that an analyser doesn't know how to handle, and your code could do the same. 

If you don't want them, just filter out those with `Fallback = true`.

aldas, 2025-04-10:
@keesverruijt ,  some smart people have blessed with the gift of having very little filtering before talking/writing in public space  https://github.com/klasyc/ScpiNet/issues/8 + other (often edited) issue threads in subject history show that.

Elmue, 2025-04-11:
The messages below PGN 0xFFFF are normally used by J1939, while NMEA 2000 uses the range above PGN 0x10000.
As on a CAN bus there is a mix of J1939 and NMEA 2000 messages you cannot assume that these messages are NMEA and contain a manufacturer code.
My example logfile shows that there are boats which do not use message 0xEF00 to send a manufacturer code.

Elmue, 2025-04-11:
> If you don't want them, just filter out those with Fallback = true.

OK. Thank you, This is the perfect solution for me. You can close this issue.

closed 2025-04-11 by keesverruijt