canboat/canboat #508: Handling of <Match> values

Elmue, 2025-04-11:
In your file the PGN 129808 exists twice:
1.) DSC Call Information
2.) DSC Distress Call Information

In the first declaration you define the second parameter as 

```
<Order>2</Order>
<Id>dscCategorySymbol</Id>
<Name>DSC Category Symbol</Name>
<BitLength>8</BitLength>
<BitOffset>8</BitOffset>
<BitStart>0</BitStart>
<Resolution>1</Resolution>
<Signed>false</Signed>
<RangeMin>0</RangeMin>
<RangeMax>253</RangeMax>
<FieldType>LOOKUP</FieldType>
<LookupEnumeration>DSC_CATEGORY</LookupEnumeration>
```

This parameter has no `<Match>` and links to a Lookup DSC_CATEGORY
When the value of the parameter is for example 110 the application will display "Urgency" as defined in the Lookup.

But if this value is 112 which the Lookup declares as "Distress" the other definition of PGN 129808 will apply where Match is 112:

```
<Order>2</Order>
<Id>dscCategory</Id>
<Name>DSC Category</Name>
<Description>Distress</Description>
<BitLength>8</BitLength>
<BitOffset>8</BitOffset>
<BitStart>0</BitStart>
<Match>112</Match>
<Resolution>1</Resolution>
<Signed>false</Signed>
<RangeMin>0</RangeMin>
<RangeMax>253</RangeMax>
<FieldType>NUMBER</FieldType>

```
But here is a problem:
The same parameter is now declared as NUMBER.
The result is that the application that uses your XML file will now display the value as Number and display "112" instead of "Distress".

Why do you define this as NUMBER now ?
The user will not understand the displayed value of 112.

In other messages with a `<Match>` you did not do this, like PGN 130816:

```
<Order>1</Order>
<Id>manufacturerCode</Id>
<Name>Manufacturer Code</Name>
<Description>Navico</Description>
<BitLength>11</BitLength>
<BitOffset>0</BitOffset>
<BitStart>0</BitStart>
<Match>275</Match>
<Resolution>1</Resolution>
<Signed>false</Signed>
<RangeMin>0</RangeMin>
<RangeMax>2045</RangeMax>
<FieldType>LOOKUP</FieldType>
<LookupEnumeration>MANUFACTURER_CODE</LookupEnumeration>
```

Here the parameter has a <Match> and is declared as LOOKUP.

keesverruijt, 2025-04-12:
Yes, this PGN is probably incorrectly defined. 

If you want to improve, thank you!

Please read https://web.archive.org/web/20170609063826/https://www.nmea.org/Assets/20130720%20%20dsc%20technical%20corrigendum%20v1..pdf
and https://web.archive.org/web/20170609021946/https://www.nmea.org/Assets/20150328%20dsc%20technical%20corrigendum%20v2.database%20version%202.100.pdf

it is a dirty mess, I agree.

Elmue, 2025-04-13:
It is worse than I thought.
The decoding of field 3 "DSC Message Address" depends on the value in field 1 (Format Specifier)

If field 1 is 120 (Individual Station) --&gt; field 3 is MMSI and from the 10 decimal digits the trailing zero must be removed to get a 9 digit MMSI.

But if field 1 is 102 (Geographic Area) --&gt; field 3 is Geo Area and from the 10 decimal digits the trailing zero must NOT be removed to maintain a 10 digit Geo Area.

The NMEA is really polulated by folks with a lack of intelligence!

To implement this correctly we would need an additional field type:
FieldType  "DECIMAL" (which must be rename to DECIMAL_9) uses their strange encoding to get a 9 digit number and
FieldType  "DECIMAL_10" which uses their strange encoding to get a 10 digit number.

Why are these folks not able to do such a simple thing as sending an ordinary integer of 32 bit which may contain any value with any number of digits as they do in all the other fields ??? They would even need 1 byte less. 

And no other message uses this strange encoding.

They need 5 bytes to send a 9 digit number. 
With unsigned integer they would need only 4 byte and could send any number up to 4294967295.
And we would not even require DECIMAL as an additional field type.
Just type NUMBER would be perfect for MMSI and for Geo Area.

And what they do in the field Expansion Data is another strange thing.

God, please throw some brains from heaven. The NMEA is in need!

TwoCanPlugIn, 2025-04-13:
While the observation that a camel is a horse designed by committee, it is worth keeping in mind that for better or worse, PGN 129808 is very much a parameter for parameter equivalent to the NMEA 0183 DSC sentence. From an implementation perspective that would have made it easy  at least for VHF/DSC radio manufacturers.

With respect to MMSI encoding, from my observations of Distress alerts and Distress Relay alerts displayed on MFD's, the encoding is as per the NMEA standard.

For example I use the following code snippet to generate PGN 129808 
```
std::string mmsiString = "2350763930";
for (size_t i = 0; i < mmsiString.length(); i += 2) {
   payload->push_back(std::atoi(mmsiString.substr(i, 2).data()))
}
```
and it results in the attached screen shot.
The actual 5 byte sequence is 0x17 0x32 0x4C 0x27 0x1E, so I think the interpretation of the standard is correct.

I'll leave it to you experts to codify the rules for the encoding of PGN 129208

BTW, A good document to accompany the NMEA standards is the ITU-R M.493-15. For example it makes it clear of the Format Specifier (field 1) is Distress (112), then the Format Category (field 2) is null. 
If the Format Specifier is Geographic Area (102) then field 3 is a 10 digit number that encodes the latitude & longitude of the bounded area. 
Both the NMEA 0183 and NMEA 2000 definitions are complex and potentially confusing. 

[image]

keesverruijt, 2025-04-14:
Given the background, I think you were trying to produce these DSC alerts. Have you seen VHFs that send this on NMEA 2000, and if so, which brands and models?

Elmue, 2025-04-16:
If this works:

`payload->push_back(std::atoi(mmsiString.substr(i, 2).data()))`

it is definitely not BCD encoding.

fakehec, 2026-08-28:
### Real on-the-wire PGN 129808 from an N2K DSC/VHF radio — plus a C-`analyzer` decode bug

Following the request for real captures: below are 4 genuine PGN 129808 messages captured off a live NMEA 2000 bus (Yacht Devices YDWG-02 gateway) originating from a **Raymarine Ray260 VHF/DSC** radio.

These are **All Ships / Safety** broadcasts (a coast-station "All Stations" safety announcement — Spanish coast station, MMSI `002241024`), **not** the Distress or Geographic-Area variants you still need. So they don't close out the format-dependent address decoding — but they do surface a concrete decoder bug.

**Bug — the C `analyzer` (v5.1.1) aborts the whole PGN on the non-distress variant:**

```
ERROR [analyzer] PGN 129808: field "MMSI of Ship In Distress" print routine did not print anything
ERROR [analyzer] PGN 129808 analysis error
```

`stdout` emits nothing for the message. For a `DSC Format = All Ships` call, the *"MMSI of Ship In Distress"* field (which belongs to the distress `Match` variant) doesn't apply, but the print routine fails on it and discards the entire PGN rather than falling through to the non-distress `DSC Call Information` variant.

**canboatjs decodes the exact same bytes correctly**, which localizes the problem to the C print path rather than the shared PGN definition:

```json
{
  "pgn": 129808,
  "description": "DSC Call Information",
  "fields": {
    "DSC Format": "All ships",
    "DSC Category": "Safety",
    "DSC Message Address": "0022410240",
    "1st Telecommand": "F3E/G3E All modes TP",
    "Subsequent Communication Mode or 2nd Telecommand": "No information",
    "Proposed Rx Frequency/Channel": "900016",
    "DSC EOS Symbol": 127,
    "Expansion Enabled": "No",
    "Time of Receipt": "11:03:00",
    "Date of Receipt": "2007.01.11",
    "DSC Equipment Assigned Message ID": 0
  }
}
```

**Raw frames** (YDWG format — `analyzer` reads these directly on stdin; 4 receptions of the same recurring safety broadcast):

```
13:53:06.071 R 11FB1004 00 3E 74 6C 00 16 29 02
13:53:06.072 R 11FB1004 01 28 64 7E 39 30 30 30
13:53:06.074 R 11FB1004 02 31 36 FF FF FF FF FF
13:53:06.075 R 11FB1004 03 FF 02 01 FF FF FF 7F
13:53:06.075 R 11FB1004 04 FF FF FF 7F FF FF FF
13:53:06.076 R 11FB1004 05 FF FF FF FF FF FF 7F
13:53:06.077 R 11FB1004 06 FC FF FF FF FF FF FF
13:53:06.077 R 11FB1004 07 FF FF FF FF FF FF 40
13:53:06.078 R 11FB1004 08 F2 B5 17 D4 34 00 00
```

(3 further identical-format receptions available; timestamps/message-IDs differ, payload is the same recurring broadcast.)

This radio emits these safety broadcasts on a schedule, so I can capture more real N2K DSC traffic if useful. I can't easily produce a genuine **Distress** or **Geographic-Area** call, which is what's still needed to finish the format-dependent address decoding — but happy to help however I can.