canboat/canboat #509: Inconsistent Min/Max Factor/Offset

Elmue, 2025-04-17:
I wrote a code that checks automatically the Min/Max values.
It take the maximum that is possible with the given bit count (e.g. 8 bit --> 255) multiplies it with Factor, adds Offset, applies the sign and checks the Min/Max values.
If found 2 errors:

PGN 127513
```
<Name>Peukert Exponent</Name>
<BitLength>8</BitLength>
<Signed>false</Signed>
<Offset>500</Offset>
<Resolution>0.002</Resolution>
<RangeMin>1</RangeMin>
<RangeMax>1.506</RangeMax>

```
Offset must be 1, not 500

__________________________________

PGN 130820
```
<Name>Transport</Name>
<BitLength>4</BitLength>
<Resolution>1</Resolution>
<Signed>false</Signed>
<RangeMin>0</RangeMin>
<RangeMax>24</RangeMax>
<FieldType>LOOKUP</FieldType>
<LookupEnumeration>ENTERTAINMENT_PLAY_STATUS</LookupEnumeration>

```
A 4 bit parameter cannot be 24. Maximum is 15.

Strange is that 

`<LookupEnumeration Name='ENTERTAINMENT_PLAY_STATUS' MaxValue='65535'>`

has 24 lookup values and is used in 3 parameters, one parameter of 4 bit, one of 8 bit and one of even 32 bit!

The lookup itself defines a maximum of 65535 which does not correspond with none of the parameters.
Something is wrong here.

To use a lookup parameter with 24 values neither makes sense in a 4 bit parameter nor in a 32 bit parameter.

sbender9, 2025-04-17:
Looks like that 130820 got broken when the lookup tables were refactored, it should not be using ENTERTAINMENT_PLAY_STATUS. 

I will do some double checking there and do a PR to fix it.

keesverruijt, 2025-04-17:
Thanks. If this code is python or some other scripting language please consider contributing the code so this doesn't happen again?

TwoCanPlugIn, 2025-04-17:
In determining maximum range do you need to consider the generally implied Data Not Available and Error values?

As an example, for an unsigned 8 bit value,  a value of 255  indicates Data Not Available and 254 means Error.

Elmue, 2025-04-18:
> In determining maximum range do you need to consider the generally implied Data Not Available and Error values?

No.
In ace of an 8 bit value the maximum is 255 and I only check for a `<RangeMax>` that is greater than 255.
If it is less (254) this does not spit out an error.

PGN 129025 defines a Min/Max range of -214.75 to 214.75 degree
The highest possible value for signed 32 bit is 7FFFFFFF, multiplied with 1E-7 = 214.7483647 lies below the maximum --&gt; no error

keesverruijt, 2025-04-19:
> A question more for Kees. What is the intent of RangeMin and RangeMax?
> 
> If they are "theoretical" limits then that can simply be deduced from the byte/bit length and sign without the need for a specific definition.

They are theoretical limits, and thus indeed redundant. Likewise for the various fields that state where in the PGN the field starts. This is also redundant as it can be computed by summing the length of the previous fields.

However, even though they are redundant they aren't useless. For example, the RangeMin and RangeMax fields are used in the production of the documentation page by applying the XSLT file. XSLT is good at transforming data, not making computations.

closed 2025-04-22 by keesverruijt