canboat/canboat #511: Fusion: Set Mute PGN (126720) was regressed by historical checkin

tomer-w, 2025-04-17:
Originally the Set Mute command was added in: https://github.com/canboat/canboat/commit/e9ae88f93cd2357d96ac00b0d366e7b79074ec7e with Proprietary ID == 17. There was bug in that PGN that it was missing the 
```
{"Unknown", BYTES(1), RES_INTEGER, false, 0, ""},
``` 
which all other PGNs which were added had.

Later it was regressed further by https://github.com/canboat/canboat/commit/73d18b13ec299594af18c9768e0d03177953adec#diff-d1afa87e4c53071c83bfad99c981d02733a24f0b6a329c8f950aef5795e06d24L1935 that wrongly changed the Proprietary ID to 23. I think this is to align with even more historical checkin - https://github.com/canboat/canboat/commit/2c03db34c4b4c44eb7f547ee634785f1b773db26 adding PGN 130820 which was now using the newly introduced FUSION_MESSAGE_ID.

I can confirm using decompiling the Fusion Wi-Fi app (which internally uses the NMEA payload to send status that the old value of 17 is the correct one.
```
    public void setMute(boolean z10) {
        FusionArrayList fusionArrayList = new FusionArrayList();
        fusionArrayList.addUInt16(17);
        if (z10) {
            fusionArrayList.addUInt8(1);
        } else {
            fusionArrayList.addUInt8(2);
        }
        ChannelAbstract channelAbstract = this.channel;
        if (channelAbstract != null) {
            channelAbstract.write(fusionArrayList.toArray());
        } else {
            setState(a.f23659v);
        }
    }
```

BTW: It also explains the that the ```UINT8_FIELD("Unknown"),``` we always have after the ```Proprietary ID``` field is not needed and the actual size of this field is 2 bytes.

so:
```FUSION_MESSAGE_ID``` should be split to 2 different lookups, one for PGN 126720 and one for 130820. The Fusion app also have 2 lists which are mostly aligned but not fully.
```LOOKUP_TYPE(FUSION_MESSAGE_ID, BYTES(1))``` should be ```LOOKUP_TYPE(FUSION_MESSAGE_ID, BYTES(2))``` This is true for both lookups.
```{"Unknown", BYTES(1), RES_INTEGER, false, 0, ""},``` should be removed from all Fusion PGN 126720
```UINT8_FIELD("A"),``` should be removed from all Fusion PGN 130820
If we are fine with this approach I can align both lookup parameter based on the actual values.

I'm far from understanding all the details of the Fusion Wi-Fi App, but based on the research this far, I'm quite sure of my findings. If anyone is interested in the decompiling work, feel free to reach out to me.

sbender9, 2025-04-17:
Are you planning on doing a PR with these changes?

I'm definitely interested in your decompilation work. I have not been able to figure everything out. 

I do use this stuff regularly to display info and control my fusion stereo. (my singnalk-fusion-plugin uses this stuff)

tomer-w, 2025-04-17:
I do plan to do a PR but I wanted to send what I understand ASAP so relevant people can chime in and let me know what they think about my findings.
I'm also remote from my boat currently and I do want to test it with real data before I submit official PR. If you can validate my PR it will be great. Do you have the mute functionality working for you?

With regards to the Fusion app, I use the jdax decompiler on the fusion APK. It produce horrible code as the app is built with Kotlin which jdax is not supporting well but it is better than nothing. I do have all the data and the different fields for both PGNs. If you are interested in specific data I can share the relevant code probably.

sbender9, 2025-04-17:
I can validate your changes. 

I just remembered that I am not using these definitions to send commands to the fusion. It's on my todo list. You can see what I send here: https://github.com/sbender9/signalk-fusion-stereo

sbender9, 2025-04-17:
I just did this PR, do you have the list of possible values for "Transport"? I've only seen the two listed. 

https://github.com/canboat/canboat/pull/510

tomer-w, 2025-04-17:
Yes, I just saw your plugin. Looks cool. I'm doing this work as I also want to have a way to control my fusion but I thought about doing it from Home Assistant integration. I really wanted to be able to play TTS from fusion but it seems limited to play mp3 only from USB and not http or NAS server. Bummer. If you have some missing functionality in your plugin. I can try to find the relevant command from the Android decompiled code.

TwoCanPlugIn, 2025-04-17:
I did some reverse engineering of this a few years ago when I had access to a neighbour's yacht.

Proof of concept plugin for opencpn which might be of interest to you.

IIRC most of the encoding/decoding is performed in twocanmedia.cpp which you can find in my [repo](https://github.com/twocanplugin/twocanplugin)

keesverruijt, 2025-04-19:
Ah, Home Assistant... fighting our Daikin heat recovery air system here.... Lovely hardware, crappy outdated Siemens PLC.

[image]

keesverruijt, 2025-04-22:
@tomer-w this can be closed now, right?

closed 2025-04-22 by tomer-w