canboat/canboat #512: Wrong negative percent values

Elmue, 2025-04-20:
Although all the following parameters are signed percent values, only positive values are allowed.
The XML file defines negative percent values to be valid which is wrong.
_____________________________________

PGN 127488 "Engine Parameters"
Parameter "Engine Tilt/Trim"

Wrong is:
`<RangeMin>-127</RangeMin>`
Correct is:
`<RangeMin>0</RangeMin>`

0% means full down, 100% means full up.
____________________________________

PGN: 127489 "Engine Parameters, Dynamic"
Parameter "Engine Load"

Wrong is:
`<RangeMin>-127</RangeMin>`
Correct is:
`<RangeMin>0</RangeMin>`

The load of the engine is never negative.
____________________________________

PGN: 127489 "Engine Parameters, Dynamic"
Parameter "Engine Torque"

Wrong is:
<RangeMin>-127</RangeMin>
Correct is:
<RangeMin>0</RangeMin>

The torque of the engine is never negative.
____________________________________

PGN: 130576 "Trim Tab / Small Craft Status"
Parameter "Port trim tab"

Wrong is:
`<RangeMin>-127</RangeMin>`
Correct is:
`<RangeMin>0</RangeMin>`

0% means full down, 100% means full up.
____________________________________

PGN: 130576 "Trim Tab / Small Craft Status"
Parameter "Starboard trim tab"

Wrong is:
`<RangeMin>-127</RangeMin>`
Correct is:
`<RangeMin>0</RangeMin>`

0% means full down, 100% means full up.
____________________________________

PGN: 127505 "Fluid Level"
Parameter "Level"

Wrong is:
`<RangeMin>-131.068</RangeMin>`
Correct is:
`<RangeMin>0</RangeMin>`

0% means empty.
Less than 0% fuel level is impossible.
_________________________________

Correcting this is important because some CAN controllers send -1 which is an invalid value.
Documentation states that all the above parameters allow only positive percent values.

A controller may send any value outside the valid range to indicate "Not available"
So this time sending -1 is not a programming error.
Sending -1 is a valid manner for all the above to indicate that the parameter is not available.
It is nonsense to display an engine torque of -1% while the engine is not running.
But the current XML file does so.

keesverruijt, 2025-04-21:
Oh, and although for some fields it does indeed make no sense, tilt/trim do actively use a -100 .. +100% range.

keesverruijt, 2025-04-22:
I'm closing this for now, feel free to re-open if you have (modern) examples of devices showing this behaviour that contradict our findings.

closed 2025-04-22 by keesverruijt

Elmue, 2025-04-22:
> Show me proof

I thought that you would be able to look up the proof yourself.
I post only one example here.

The NMEA document from that I took a screenshot, has been leaked by a Chinese guy who was so kind to provide us with information that NMEA wants to keep secret. We can consider this document public now because it has been published 4 years ago. So everybody who wants to have it, has already downloaded it. It is a very naive idea that in times of internet companies or organizations can keep information secret.
Just think about the emails from Hillary Clinton or the Panama papers. The truth comes to the light some day and so will further NMEA documents. Once leaked we can consider them as public.

[image]

You can lookup all the others PGNs on your own.
All what I wrote above is proved by the same document.
Do you need more proof ?

> feel free to re-open

I cannot because Github says I have no permission.

keesverruijt, 2025-04-22:
Yes, I do. I already made the argument that this 1.3 or whatever version of the documentation is NOT permissible and I do not want to read it and it is OUTDATED.

For all these percentage fields the products that I trust from brands such as Furuno and Antisense (especially the latter as they are heavily involved in creating the NMEA standard) produce signed values for Unknown (e.g. 0x7f in this 8 bit integer case). Therefore the fields are a signed 8 bit value.

For TRIM I can put in a negative trim value in percentage in the emulator that I now have and my Antisense NGT-1 will show a negative value.

I can't help it if you don't like it.

Elmue, 2025-04-22:
Nobody ever doubted that these percent values are signed.
I never wrote that they are not.
The question is only if negative percent values are allowed or not.
And for all of them the document, that you consider outdated, states that the values should be only positive.

And for example for Engine Load and Engine Torque negative values will not make sense.