canboat/canboat #513: Contradicting Manufacturer code

Elmue, 2025-04-21:
In the logfile susteranna2020.raw I find a controller at source address 1B that sends a contradicting manufacturer code.
I suspect that something is wrong here.

[image]

keesverruijt, 2025-04-22:
All the same company, after some takeovers and mergers. Even more mergers have happened since, so I wouldn't be surprised if we see a B&G product produce CZone, Mastervolt and Brunswick PGNs in a few years. 

I think the correct conclusion is that a NMEA product can send company specific PGNs that were invented by any company, as long as it knows how.

BTW -- screenshot of your product, I would not use "NMEA (Match)" but "Company Specific" as wording. And for a low level analyzer like yours, as a customer, I would want to know what bytes form the fields. So either print those bytes (in an extra column or to the right of the translated value) or show little marks below the data bytes -- maybe a coloured stripe?

closed 2025-04-22 by keesverruijt

Elmue, 2025-04-22:
> All the same company, after some takeovers and mergers.

I considered this possibility before posting this issue.
But when I Googled for "B & G" and "Simrad" I could not find any connection between these companies.
Both are selling their products under their own name on several webpages.
There are products from B & G and others from Simrad.

And in the latest version of the Manufacturer list they are still listed as separate companies:
https://www.isobus.net/isobus/manufacturerCode

Simrad
381 --> B & G (Romsey, UK)
1857 --> Simrad (Egersund Norway)

I could not find that they have fusioned.
Do you have any link about this ?
____________________________

> I would not use "NMEA (Match)" but "Company Specific" 

Good idea. I changed that,

> or show little marks below the data bytes -- maybe a coloured stripe?

This would not work for bytes that contain multiple parameters.
For example 2 bytes containing 11 bit Manufacturer code + 2 bit Reserved + 3 bit Industry Group.
The coloured stripes would be only a few pixels long.

My software is made for workshops that repair motorbikes, cars and in the next version also J1939 trucks, buses and NMEA yachts.
The people working there are mostly interested in the DTC failure codes and some more engine parameters.
But showing them binary data will only confuse them.

But what I can do is to show a tooltip with more information when the mouse is over a parameter.

nohal, 2025-04-22:
> Do you have any link about this ?

You mean something like https://www.navico.com/our-brands?

keesverruijt, 2025-04-22:
And Navico has been bought by Brunswick, so: https://www.brunswick.com/our-brands

Elmue, 2025-04-22:
Interesting

[image]