canboat/canboat #521: PGN 130817 for Navico is surely wrong

Elmue, 2025-04-24:
There are 2 declarations of PGN 130817.
1.) Lowrance: Product Information
2.) Navico: Product Information

In logfile "susteranna 2020" I see Lowrance data that makes sense:

[image]
__________________________________

But all the others which are Navico are totally wrong.

Logfile "susteranna actisense serial" and "merrimac 2022":

[image]

As you see the message has only 7 bytes, but 4 fix length strings with 32, 10, 32, 32 characters are expected.
The crippled data for "Model" is surely not ASCII.

[image]

[image]

[image]

All these messages are only 7 byte long and surely do not contain 4 fix length strings for model, firmware version, firmware date and firmware time.

keesverruijt, 2025-04-28:
First of all, closer inspection of `samples/susteranna2020.raw` shows that this contains inconsistencies. It is in "all on one line" format with fast PGNs already converted into single lines, for example:

    2020-08-22T13:52:36.951Z,3,127251,128,255,8,00,3e,6f,fe,ff,00,ff,ff
    2020-08-22T13:52:36.951Z,3,129029,127,255,43,00,40,48,e0,3a,c7,1d,40,6f,16,41,06,28,61,07,a8,90,61,96,1b,0f,c1,00,d0,fb,01,00,00,00,00,00,24,fc,00,3c,00,5a,00,34,12,00,00,00

But the hardware/software combination, most likely an NGT-1, did not do this correctly for all company specific PGNs in range 0x1FF00-0x1FFFF, where it actually produces one line per frame.

So what you call "Valid FastPacket Data" in your image above, should really be all be shown on one line.

My analyzer parses this file correctly if you use the `-format PLAIN_MIX_FAST` argument.

The `samples/merrimac2022.raw` file is consistent: all fast PGNs are on one line.

But, as you point out, it contains PGN 130817 messages (starting with `13 99` and just 7 bytes in total) that do not match the definition in the PGN database. This 7 byte version can also be found in `samples/susteranna-actisense-serial.raw`.

It looks a little like the key/value data in "PGN 130824 - B&G: key-value data" to be honest.

It's probably best if we update the `13 99` version of PGN 130817 accordingly.

Elmue, 2025-04-30:
I also have the impression that the software which creates these logfiles, is very buggy.
I see in the same logfile FastPackets that are decoded into one line and other FastPackets that are not.

I suppose there is a bug in the software that they implemented wrongly which PGNs are fast and which are single packets.

But there is no logfile in your folder "samples" that contains valid data for Navico: Product Information.
The logfiles that exist are all corrupt.

Do you have more logfiles than these were you did find this message ?
Otherwise where does the XML specification for Navico come from ?

keesverruijt, 2025-05-01:
> I also have the impression that the software which creates these logfiles, is very buggy. I see in the same logfile FastPackets that are decoded into one line and other FastPackets that are not.
>
> I [...]

Yes, and it is one of the downsides of logging data from official NMEA 2000 gateways, which are supposed to coalesce fast packets themselves. Logging CANdump style where every frame is logged is much better over the long term as it contains no interpretation.

Unfortunately the logfiles don't clearly state what device was used to defragment Fast PGNs and what firmware or software version was used.

> Do you have more logfiles than these were you did find this message ? 

No. But once the boat is back in business I may be able to generate more.

> Otherwise where does the XML specification for Navico come from ?

You are so funny. I just made up what made sense at the time. The Navico definition was just a copy/paste error I think.

The trouble with your way of thinking is that you want every tree in the forest to stand up. In real life, a forest contains trees that have been blown over, are dead, or just grow crooked.
You will have to live with this, I'm afraid.

Elmue, 2025-05-02:
> The trouble with your way of thinking is that you want every tree in the forest to stand up.

There is no trouble with my thinking.
It is obvious that all these messages which show parameters with names A, B, C, D, E, F are not yet reverse engineered.

> In real life, a forest contains trees that have been blown over, are dead, or just grow crooked.

I would not compare these messages with trees that are blown over.
They are rather like a seed that some day wants to become a tree.