canboat/canboat #522: Route and Waypoint Service - PGN's 130064 - 130074

TwoCanPlugIn, 2025-04-24:
A  few years ago added support in my OpenCPN plugin to import/export waypoints with B&G MFD's using PGN 130074.

Currently have access to a Raymarine Axiom MFD, which in First Responder mode supposedly can import/export routes and waypoints over NMEA 2000. Whilst reverse engineering, and based on observations, especially for PGN 130074, the current canboat definitions are incorrect for all of the Route & Waypoint Service PGN's. (130064--130074)

All of the initial fields of these PGN's, such as Route Point Sequence (rps), numbers of items, numbers of valid items, database, route & waypoint Id's are all 2 bytes in length (unsigned shorts).

I've attached my sample code for PGN 130064, which is parsed correctly with Actisense Reader.

Hopefully you'll understand the initial field lengths. If you need samples for the remaining PGN's feel free to ask.

```
// Routes - Database List
void EncodePGN130064(std::vector<byte>* payload) {
	payload->clear();

	unsigned short startRPS = USHRT_MAX;
	payload->push_back(startRPS & 0xFF);
	payload->push_back((startRPS >> 8) & 0xFF);

	unsigned short items = 2;
	payload->push_back(items & 0xFF);
	payload->push_back((items >> 8) & 0xFF);

	unsigned short databases = 1;
	payload->push_back(databases & 0xFF);
	payload->push_back((databases >> 8) & 0xFF);

	unsigned short databaseId = 1;
	payload->push_back(databaseId & 0xFF);
	payload->push_back((databaseId >> 8) & 0xFF);

	std::string databaseName = "SQL Server";
	payload->push_back(databaseName.size() + 2);
	payload->push_back(0x01); // 1 indicates ASCII encoding
	for (size_t i = 0; i < databaseName.size(); i++) {
		payload->push_back(databaseName.at(i));
	}

	time_t seconds;
	seconds = time(NULL);

	unsigned short daysSinceEpoch = seconds / (86400);
	unsigned int secondsSinceMidnight = (seconds - (daysSinceEpoch * 86400)) * 10000;

	payload->push_back(secondsSinceMidnight & 0xFF);
	payload->push_back((secondsSinceMidnight >> 8) & 0xFF);
	payload->push_back((secondsSinceMidnight >> 16) & 0xFF);
	payload->push_back((secondsSinceMidnight >> 24) & 0xFF);

	payload->push_back(daysSinceEpoch & 0xFF);
	payload->push_back((daysSinceEpoch >> 8) & 0xFF);

	// waypoint resolution & reserved
	byte waypointResultion = 3;
	payload->push_back((waypointResultion & 0x0F) | 0xF0); // waypoint resolution is lower 4 bits

	unsigned short nRoutes = 2;
	payload->push_back(nRoutes & 0xFF);
	payload->push_back((nRoutes >> 8) & 0xFF);

	unsigned int nWaypoints = 69;
	payload->push_back(nWaypoints & 0xFF);
	payload->push_back((nWaypoints >> 8) & 0xFF);
	payload->push_back((nWaypoints >> 16) & 0xFF);
	payload->push_back((nWaypoints >> 24) & 0xFF);

	unsigned int nBytes = 1024;
	payload->push_back(nBytes & 0xFF);
	payload->push_back((nBytes >> 8) & 0xFF);
	payload->push_back((nBytes >> 16) & 0xFF);
	payload->push_back((nBytes >> 24) & 0xFF);

	// Second DB
	databaseId = 2;
	payload->push_back(databaseId & 0xFF);
	payload->push_back((databaseId >> 8) & 0xFF);

	databaseName = "DB2";
	payload->push_back(databaseName.size() + 2);
	payload->push_back(0x01); // 1 indicates ASCII encoding
	for (size_t i = 0; i < databaseName.size(); i++) {
		payload->push_back(databaseName.at(i));
	}

	seconds = time(NULL);

	daysSinceEpoch = seconds / (86400);
	secondsSinceMidnight = (seconds - (daysSinceEpoch * 86400)) * 10000;

	payload->push_back(secondsSinceMidnight & 0xFF);
	payload->push_back((secondsSinceMidnight >> 8) & 0xFF);
	payload->push_back((secondsSinceMidnight >> 16) & 0xFF);
	payload->push_back((secondsSinceMidnight >> 24) & 0xFF);

	payload->push_back(daysSinceEpoch & 0xFF);
	payload->push_back((daysSinceEpoch >> 8) & 0xFF);

	// waypoint resolution & reserved
	payload->push_back(0xFF);

	nRoutes = 32;
	payload->push_back(nRoutes & 0xFF);
	payload->push_back((nRoutes >> 8) & 0xFF);

	nWaypoints = 96;
	payload->push_back(nWaypoints & 0xFF);
	payload->push_back((nWaypoints >> 8) & 0xFF);
	payload->push_back((nWaypoints >> 16) & 0xFF);
	payload->push_back((nWaypoints >> 24) & 0xFF);

	nBytes = 16371;
	payload->push_back(nBytes & 0xFF);
	payload->push_back((nBytes >> 8) & 0xFF);
	payload->push_back((nBytes >> 16) & 0xFF);
	payload->push_back((nBytes >> 24) & 0xFF);

}
```

[image]

tkurki, 2025-04-25:
Thanks for sharing, added a tracking issue on Signal K side as this would be valuable also there.

keesverruijt, 2025-04-25:
@TwoCanPlugIn can you attach some logfiles, preferably in something ingestible to analyzer?

(I really should add .ebl format support to canboat... can't be that hard.)

aldas, 2025-04-25:
Tricky parts from `.ebl` is timestamp related parts if you are also interested in timings.  

About 2 years ago I asked Actisense about that format and answer was
> We don't provide support for developers trying to interpret EBL files for themselves. Also, we can't guarantee that the format will not change arbitrarily in the future. The format of EBL files is completely proprietary and unpublished.

keesverruijt, 2025-04-26:
@TwoCanPlugIn I was looking for the rest of your code in your plugin, but is this maybe not in Github yet? I cannot even find EncodePGN130064 there -- I see the Decode versions of 130065 and 130074 but 130065 seems to use 1 byte IDs which you are saying is incorrect?

TwoCanPlugIn, 2025-04-26:
Hi Kees,

No I haven't pushed my latest stuff up there yet.

Trying to fix a few other unrelated bugs as well as the continual problems with changes to the circleci build process.

IIRC whatever PGN 130074 stuff up there is correct as I know that I could receive newly created waypoints from a B&G MFD.

Anyway find attached my quick & dirty POC that I used to generate the EBL files that I sent previously. You should be able to get an idea of both the Actisense EBL format as well as the fields for PGN's 130064 - 130074. It's inelegant and kludgy, a lot of cut-n-paste from the plugin but hopefully you'll be able to understand it.

I'm reasonably confident the PGN's display correctly in the Actisense Reader software.

Regards.

// ebl-writer - write data using Actisense RAW Protocol

// Compilation on Windows cl ebl-writer.cpp /std:c++17 /EHsc

#define WINDOWS_LEAN_AND_MEAN
#define _USE_MATH_DEFINES
#include <windows.h>

#include <string>
#include <vector>
#include <iterator>
#include <cmath>
#include <ctime>
#include <unordered_map>
#include <signal.h>

static volatile bool canThreadIsAlive = true;
unsigned int sequenceId;
HANDLE fileHandle;

// ASCII Control Chracters
const byte DLE = 0x10;
const byte STX = 0x02;
const byte ETX = 0x03;
const byte ESC = 0x1B;
const byte BEMSTART = 0x01;
const byte BEMEND = 0x0A;
const byte EBL_VERSION = 0x01;
const byte EBL_TIME = 0x03;
const byte EBL_ELEMENT = 0x04;

// Actisense NGT-1 commands
const byte N2K_TX_CMD = 0x92;
const byte N2K_RX_CMD = 0x93;
const byte NGT_TX_CMD = 0xA1;
const byte NGT_RX_CMD = 0xA3;

#define CONST_HEADER_LENGTH 4
#define CONST_PAYLOAD_LENGTH 8
#define CONST_FRAME_LENGTH CONST_HEADER_LENGTH + CONST_PAYLOAD_LENGTH
#define CONST_GLOBAL_ADDRESS 255

#define CONST_MAX_MESSAGES 100
#define NOT_FOUND -1

#define RADIANS_TO_DEGREES(x) ((x) * 180.0f / M_PI)
#define DEGREES_TO_RADIANS(x) ((x) * M_PI / 180.0f)
#define CONST_KELVIN -273.15f
#define CONVERT_MS_KNOTS 1.94384
#define CONVERT_METRES_NAUTICAL_MILES 0.000539957

#define HEADING_TRUE 0
#define HEADING_MAGNETIC 1

#define TIME_SOURCE_GPS 0
#define TIME_SOURCE_GLONASS 1
[... 1770 lines cut ...]
	for (size_t i = 0; i < waypointName.size(); i++) {
		payload->push_back(waypointName.at(i));
	}

	latitude = -34.8 * 1e7;
	payload->push_back(latitude & 0xFF);
	payload->push_back((latitude >> 8) & 0xFF);
	payload->push_back((latitude >> 16) & 0xFF);
	payload->push_back((latitude >> 24) & 0xFF);

	longitude = 145.4 * 1e7;
	payload->push_back(longitude & 0xFF);
	payload->push_back((longitude >> 8) & 0xFF);
	payload->push_back((longitude >> 16) & 0xFF);
	payload->push_back((longitude >> 24) & 0xFF);

}

// Routes - Waypoint Name
void EncodePGN130068(std::vector<byte>* payload) {

	unsigned short items = 3;

	unsigned short waypoints = 100; 
	payload->push_back(w

commit 59581f24 (2025-04-28): Fix #522
closed 2025-04-28 by keesverruijt