canboat/canboat #523: Add Antisense .ebl format support

keesverruijt, 2025-04-25:
It would be useful to be able to analyse .ebl files. Can't be too hard.

aldas, 2025-04-25:
Tricky parts from .ebl is timestamp related parts if you are also interested in timings.

I you are insterested then you could look my try at it at here https://github.com/aldas/go-nmea-client/blob/75bc78b9f6f828f93bc7c51ce4cd87e738f89c12/actisense/eblreader.go#L12

and test data from Actisense W2K-1 device https://github.com/aldas/go-nmea-client/blob/75bc78b9f6f828f93bc7c51ce4cd87e738f89c12/actisense/testdata/actisense_w2k1_bst95.ebl (I do not mind if you want to include these in canboat repository)

Maybe this helps - this was question that I asked Actisense about timestamps.
```txt
Hi,  if it  is not a huge secret - how timestamps are handled in EBL files?  From some experiments with EBL Reader I can deduce that

In EBL reader "Message Details" frame  "Timestamp XXXXX milliseconds" is 2nd and 3rd bytes in data frame in file (little endian). 
In EBL reader "Timestamp" column values are probably derived from special data frame/row in EBL file. Which is first in file and seems to repeat when data rows roll over. This is frame which starts with `1b 01` + `03` and has 8 bytes. For example `1b01 03ff 2975 2400 0000 001b 0a` seems to result `00:01:01.165` (decimal 611658239 -> 61seconds, 165millisec, 8239microsec). So I assume it is microseconds since something (start of day/boot). Seems that all 8 bytes are related to time but does not seem to be unixtimestamp. I assume it is maybe microseconds since device(microcontroller) start?
So my question is: how to interpret rows/frames in EBL file that have type `03`. How these 8 bytes should be read to get "global" time for next data rows in EBL.

For example:  

File starting with `1B 01 03 00 10 E7 A7 84 83 D9 01 1B 0A` is somehow "21:16:16.000" as timestamp in GUI.  8Bytes `00 10 E7 A7 84 83 D9 01` is   uint32 (Little endian): 2816937984  and uint64 (LE) 133282269760000000
How 133282269760000000 should be interpreted to get "21:16:16.000" as timestamp (lets assume data message 2nd and 3rd bytes are 0x0 0x0 - so no offset).

Currently we can data out of EBL files to be analyzed but for some use-cases (sea trials) we want to know for example how long time it took to reach certain heading etc.
```

keesverruijt, 2025-04-25:
Thanks Martti.

It's a pity these companies don't realise that they are leaving money on the table by being needlessly secretive. I'm sure they would sell more hardware if the software spec was open.

aldas, 2025-04-25:
I have had some success with them - they actually provided me with some documents for formats that W2K-1 can be configured to use (n2k ascii and N2K Actisense). But these were advertised in their manual as something that they provide - I emailed them and got the formats.  If you are interested - code about these formats are also in that repository. 

I think they want to funnel people more to their cloud offering https://actisense.com/actisense-cloud/ which can decode these files.

TwoCanPlugIn, 2025-04-25:
Re EBL,

EBL files simply encapsulate Actisense Raw messages with two subtle differences. Keep in mind that Actisense Reader is WIndows only, so that gives a subtle hint....

Firstly EBL files include headers; IIRC a header identifier (differentiates a time stamp from a version number). The time stamp is an 8 byte field containing a Windows FileTime. The version number, IIRC is 2 bytes in length. 

The second thing to note is that the headers are delimited by the Escape Character (0x1B) followed by a Start/End character (0x01 and 0x0A respectively). And similar to Actisense Raw data where the DLE character must be escaped, so too the ESC character.

Eg.
ESC, BEMSTART, TIMESTAMP ID, TIMESTAMP, ESC, BEMEND, ESC, BEMSTART, VERSION ID, VERSION NUMBER, ESC, BEMEND, ESC, BEMSTART, TIMESTART ID, TIMESTAMP, ESC, BEMEND, DLE, STX, ACTISENSE RAW DATA, DLE, ETX

Decoding Matti's example header

1B 01 ESC, BEMSTART

03 TIMESTAMP HEADER ID

00 10 E7 A7 84 83 D9 01 Windows FileTime

1B 0A ESC, BEMEND

If you convert the Windows Filetime to an integer (remembering the endianess) you can paste it into a tool such as[ this](https://www.epochconverter.com/ldap) and display the same time as shown in the Actisense Reader software.

Re the other Actisense formats (ASCII RAW, ASCII N2K), they do provide the documention and they were provided to the OpenCPN team in order to add support for those formats in addition to Actisense Raw in recent versions of OpenCPN.

commit 9811d369 (2025-04-28): Fix #523: actisense-serial can now read .ebl files
closed 2025-04-28 by keesverruijt