canboat/canboat #525: Sync with latest publicly known sources

keesverruijt, 2025-05-01:


keesverruijt, 2025-05-01:
@sbender9 regarding our field name discussions, I found that I had, for some reason, misnamed the 3rd corner field as "corner 1" instead of "corner 2" in some obscure AIS messages (PGNs 129806 and 129087). I decided to fix this, please give feedback if this is breaking for you or not.

sbender9, 2025-05-01:
We don't use that, so won't affect sk.

Elmue, 2025-05-02:
I'll need some time to check all this.

Elmue, 2025-05-03:
I checked the latest XML file.
I begin with the small things:
______________________________________

You write for String LZ: "The character encoding is believed to be UTF-8."
I wrote in my last post under issue 498 that this is not only "believed". 
I found a string in the logfile that **confirms** that they definitely send UTF-8 strings.
______________________________________

The lookup OFF_ON_CONTROL still says "On", "Off" rather than "Turn On/Off" or "Enable/Disable"
It is only used in "Switch Bank Control" to turn on/off the 28 switches.
______________________________________

Where does your information come from which PGNs are fastpacket?
I found a contradicting information.

CSS Electronics sells the entire NMEA data base for 1200 € so I consider them as a reliable source.
They uploaded a Python sample code where they treat PGN 127751 as fastpacket.
https://github.com/CSS-Electronics/api-examples/blob/master/examples/data-processing/utils.py

Search for "res_id_list" under "nmea".
I don't know what is correct.
Do you have any reliable source with all PGNs that are fastpackets ?
_____________________________________

You did a lot of updates in the last version!
Excellent.
Great work.

[image]

_____________________________________

I started to verify the last XML file manually but I found this work too tedious.
It is very boring and it is very easy to overlook an error.
There are 1107 parameters to be checked.

Therefore I wrote a code that extracts the PDF content and converts it into your XML format.
Now I have the entire PDF as XML and can compare it with your XML file.

What my code does:
------------------
It extracts for the parameter group: 
PGN, ID, Description, Priority, Type, FieldCount, Length

And iIt extracts for the parameters: 
Order, ID, Name, BitLength, BitOffset, BitStart, Resolution, Signed, RangeMin, RangeMax, FieldType, TypeInPdf

It handles String_FIX and String_LAU.

It converts units with their Factor/Min/Max like:
"cu- m/hr" --> "L/h"      
"Pi" and "2Pi" --> float values
"1/32 x 10E-3" --> float value
"C" --> "Ah"
"minute" --> "s"
"deg K" --> "K"
"sec(\")" --> "deg"

It detects errors in the PDF:

1.)
PGN 129550: Field 6 is missing.
But this is only a typo in the document.
If you check the last document from last year (2024), there are also not more fields:
https://web.nmea.org/External/WCPages/WCWebContent/webcontentpage.aspx?ContentID=189

2.)
PGN 127503 and PGN 127504 field 6 have a contradiction:
Unsigned type uint16 uses a negative RangeMin value.

What my code is not doing:
---------------------------
It does not handle repeated parameters (RepeatingFieldSet1xxx).
It does not extract the values for lookups (LookupEnumeration).
It does not automatically assign the types: DATE, TIME, DURATION, LENGTH, DISTANCE, GEOGRAPHICAL_COORDINATE.
It does not automatically assign the PhysicalQuantity

Here I upload the XML file.
It has 16200 lines.
It is 100% auto-generated.
You can compare it with yours.
I use Araxis Merge for this, an old but very useful program.
If you are interested I can send you also the source code.

[NMEA Extracted.zip](https://github.com/user-attachments/files/20025127/NMEA.Extracted.zip)

Some examples:

The Vehicle Identification Number is String_LAU:

[image]

This field is a lookup:

[image]

Here the unit, signed, Min and Max are wrong:

[image]

Here a contradiction in the PDF was detected:

[image]

I would recommend to use as few as possible abbreviations.
The original document says "Sequence ID" which you abbreviated to "SID".
The same applies to SOG and COG and others.
May be for you these abbreviations are your daily bread.
But there will always be newcomers who see this for the first time and it would help them to avoid abbreviations.

Here you abbreviated "Ship Length" to "Length".
"Length" may be anything, while "Ship Length" is clear.

[image]

And here you added a comment, but depending on the application that uses your XML file this comment may never be displayed. And I think the comment is not necessary if the name is clear.

[image]

"Time elapsed (from) or to-go to mark"
means that a positive time indicates the time until a destination point is reached.
And it may be negative for a time that has elapsed since an event in the past.

You abbreviated this as "Time to mark" which does not reflect the intention of this parameter.

I would recommend:
"Time to go to mark (positive) / Time elapsed (negative)"

_____________________________________

P.D.
What is the purpose to add a Reserved field to the end of a message behind the last parameter?

keesverruijt, 2025-05-04:
> STRING_LZ

Yes, this is recently been verified as UTF-8. I will fix this.

> OFF_ON_CONTROL still says "On", "Off" rather than "Turn On/Off" or "Enable/Disable"

Yes, I find this clearer. If something is already a command then "On" is a command. 
 
> Where does your information come from which PGNs are fastpacket?: 

Heuristics originally, anything over 8 was fast packet; later we discovered some ranges contain only single or fast packet framed PGNs. If you think about it, the company specific ranges have to be -- otherwise you cannot discern your own brand PGNs from other brands. 

It's a pity the NMEA did not put single frame and multi frame all in different ranges. So for some PGNs that are 8 or fewer data bytes the only definitive source is the NMEA database, which we cannot access and then distribute our knowledge of. So we have changed a lot of single/multi frame assignments over time, as more data became available. 

The best way to determine single/multi frame is to use a low level CAN interface instead of a NMEA gateway and look at the data on the bus.

> Field names are not identical

Yup. Some entries in the data pre-date ALL disclosures of the NMEA. They started providing a list of field names only "recently" (2019). For upwards compatibility with downstream code we stopped changing field names. Thus, we'll just have to live with them. Feel free to create a PR that contains the longer names (in the `pgn.h` file) as the description, or wait for the ability to define a separate technical name from the end user name.

keesverruijt, 2025-05-04:
> What is the purpose to add a Reserved field to the end of a message behind the last parameter?

For single frame messages to round out the length to 8 bytes (64 bits). For fast frame messages to round out the length to a full byte.

Elmue, 2025-05-04:
> make an XSLT processor 

This is not necessary.
I can compare the XML files without changing anything, as you see in my screenshots.

> showed diffs but in a not as nice format. 

I have no idea what you mean with this.
Could you post a screenshot ?

I have an old Araxis version that works perfectly.
If you want I can send it to you.
Send me an email.

As the auto-generated XML file starts with PGN 126992 you must delete in you XML file all PGNs before 126992  otherwise Araxis Merge will not be able to compare the files - too many differences.

I can also send you the sourcecode and you modify the output format as you like it.
But in my opinion this is not necessary.
The XML file that I uploaded can be compared directly with yours.

____________________________

> The best way to determine single/multi frame is to use a low level CAN interface instead 
> of a NMEA gateway and look at the data on the bus.

And what about installing Actisense NMEA Reader and connecting it to a CAN device emulator ?
The emulator broadcasts packets to CAN bus and you see how Actisense decodes the packets.
Then you manipulate the data bytes and see how the displayed values change.
Do you have a NGT-1 interface ?

Elmue, 2025-05-09:
@Kees:
> It's a pity the NMEA did not put single frame and multi frame all in different ranges.

I did a reverse engineering of an NMEA software (I will not mention which one here) and I found an interesting logic that they use to distinguish between single packets and fast packets:

For all known PGNs (the PGNs which the software can decode) they have a lookup table where for each PGN is defined separately if it uses single or fast packets.

But interesting is the fallback mechanism which they use for unknown PGNs:

**Only** for the following ranges they display that it is an "Unknown Fast Packet PGN":
```
0x1F000 until 0x1F01A
0x1F100 until 0x1F11A
0x1F200 until 0x1F21A
0x1F300 until 0x1F31A
-----------------------
0x1F500 until 0x1F51A
0x1F600 until 0x1F61A
-----------------------
0x1F800 until 0x1F81A
0x1F900 until 0x1F91A
0x1FA00 until 0x1FA1A
0x1FB00 until 0x1FB1A
0x1FC00 until 0x1FC1A
0x1FD00 until 0x1FD1A
0x1FE00 until 0x1FE1A
-----------------------
0x1FF00 until 0x1FFFF  (entire range)
```

**ATTENTION:**
That these are **reserved** for fast packets does not mean that all messages in this range **must** be fast packets.
They may also use single packets in these ranges,

On the other hand for the following ranges they display that it is an "Unknown Single Packet PGN"
```
0x1F400 until 0x1F4FF
0x1F700 until 0x1F7FF
```

After finding this algorithm I compared it with my list of fast packet PGNs that I use currently and very interestingly not even one  fast packet PGN exists where the low byte is greater than 0x1A (except in the last range 1FFXX).
And also not even one fast packet PGN exists in the range 0x1F4XX or 0x1F7XX.
This confirms that this rule in deed exists.
So it seems that the NMEA has made this secret agreement with the manufacturers.
                
I find it interesting that there are only very few PGNs reserved for fast packets.
It seems that the NMEA wants to encourage the manufacturers to use single packets wherever possible, which makes much sense to reduce bus traffic. 

Several manufacturers seem to have no conscience that CAN bus has been designed to send small packets.
The design of String_LZ always wastes one useless byte: You need either a length byte or a zero termination, but never both.
And a message like in logfile pgn129808.raw where they send a String_LAU like this:
12,01,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF
(length = 18 bytes full of FF) shows a firmware developer without any conscience of reducing useless bus traffic.

May be you add this to chapter "PGN ranges" on your page canboat.html

russkel, 2025-10-16:
> And what about installing Actisense NMEA Reader and connecting it to a CAN device emulator ?

Could you please elaborate on this CAN device emulator?