canboat/canboat #528: GLL missing in current version

maikpfeil, 2025-05-02:
Hello, 

I use Openplotter which provide binary for CANboat version v5.0.3. In this version I get xxGLL exported from n2kd via TCP to my third party app.

Example: $JGGLL,3.0873,N,8646.3287,W,,A,D*47

However with latest pull from GitHub which shows me CANboat version v5.2.0 after compiled candump2anlayzer/analyzer & n2kd, I don't receive this message anymore. 

Switching back to Openplotter version it appears again.

Do you have any idea what change could caused this?

keesverruijt, 2025-05-02:
Can you try the 5.2.0 tag please instead of head?

maikpfeil, 2025-05-02:
Only found tag until v5.1.3 but will try with this version now.

keesverruijt, 2025-05-02:
Sorry I meant 5.1.3

maikpfeil, 2025-05-02:
Compiled v5.1.3 and works well, like v5.0.3.

keesverruijt, 2025-05-04:
Hmm I can't reproduce (yet), see branch [n2kd_nmea0183_tests](https://github.com/canboat/canboat/tree/n2kd_nmea0183_tests). 

Can you condense some output from analyzer that produces GLL output for you with 5.1.3 or 5.0.3 and does not with master?

N2Kd only uses PGN 129029 to generated a GLL message, this hasn't changed.

keesverruijt, 2025-05-04:
I did just now fix a small bug, the time field in GLL wasn't filled but I think this has been bad for a long time, at least v5.0.0.

maikpfeil, 2025-05-05:
I tried several branches and this is my summary.

############ Initial test with v5.0.3					----> Result GOOD $JGGLL,136.5050,S,9007.0850,W,,A,D*50

############ test with branch n2kd_nmea0183_tests	----> Result GOOD $JGGLL,136.5590,S,9007.2746,W,010302,A,D*53

############ test with branch master (6.0.0)			----> Result GOOD $JGGLL,136.6198,S,9007.5068,W,,A,D*50

############ test with branch master and reverted last two commits	----> Result GOOD $JGGLL,136.7322,S,9007.9183,W,,A,D*5A

--> All working branches showing GLL! Maybe I missed something some days ago, I don't know, can't explain. 

However I'm really happy about the time fix, would be happy to see this merged in master soon.

Please see attached log file for all tests in detail, if you don't see anything I believe we can close this issue as not reproducable.

[canboat_GLL_log.txt](https://github.com/user-attachments/files/20030914/canboat_GLL_log.txt)

closed 2025-05-08 by keesverruijt