canboat/canboat #529: Issue with FIELDTYPE_LOOKUP / KEY_VALUE

Elmue, 2025-05-05:
I try to implement FIELDTYPE_LOOKUP / KEY_VALUE.
But it is currently not possible.

PGN 130824 has Param 4 = FIELDTYPE_LOOKUP and Param 6 = KEY_VALUE
PGN 130845 has the same scheme.
PGN 130846 has the same scheme.

But PGN 130833 does not match this scheme:
Param 1 = Manufacurer Code   (11 bit)
Param 2 = Reserved                  (2 bit)
Param 3 = Industry Group        (3 bit)
Param 4 = FIELDTYPE_LOOKUP (12 bit)
Param 5 = Length                      (4 bit)
Param 6 = Reserved                  (8 bit)
Param 7 = LOOKUP                  (8 bit)
Param 8 = Short Name             (64 bit)
Param 9 = Long Name              (128 bit)

Here the KEY_VALUE parameter is missing.
To decode this message I have to know to which field the FIELDTYPE_LOOKUP refers.
But the XML file does not give me this information.
_____________________________________

A second problem is that all these messages have a Length field for the Value field, but this field is marked as NUMBER.
How do I know which field contains the length for the value ?
Currently it is not possible to find out.
We need a new field type: VALUE_LENGTH instead of NUMBER.

So to decode these messages, **each** message should contain these 3 fields:
- a parameter FIELDTYPE_LOOKUP
- a parameter VALUE_LENGTH 
- a parameter KEY_VALUE

Or do you simply ignore the Length field and take the length that is stored in 'Bits=16' in the `<LookupFieldTypeEnumeration>` ?
_____________________________________

I also don't understand why PGN 130833 has a Length field while all the fields in the message have a fix BitLength.
There is no field with variable length.

aldas, 2025-05-08:
If it is not too much to ask - all these API changes. including Enums - it would be really helpful for downstream projects if these are mentioned in changelog as bold breaking change titled block. If you have handwritten you enums and did not create some kind of validation functionality against canboat XML these are things that you find out in runtime when enum value what you do not expect shows up or something that you expect is not there anymore.

keesverruijt, 2025-05-08:
@aldas I expanded CHANGELOG.md -- the changes aren't really that much for projects that use the main PGNs. The above only covers Simrad and B&G PGNs that are very flaky anyway. Wouldn't rely on them just yet. 

In fact I found some glaring errors in 130845 (Simnet key/value) and I will need to spend a few days this summer getting to the bottom of them. It's just mindblowingly slow work.

sbender9, 2025-05-08:
We are using these in Signal K

https://github.com/SignalK/n2k-signalk/blob/master/simrad/130845.js

keesverruijt, 2025-05-08:
> We are using these in Signal K
> 
> https://github.com/SignalK/n2k-signalk/blob/master/simrad/130845.js
> 
> https://github.com/sbender9/signalk-n2k-displays

Not affected as far as I can see, it's only the metadata in the XML that is affected "what is the type of the field". Your code doesn't care.

htool, 2025-05-08:
@keesverruijt  I'm trying to find what PGN 129833 is/was doing, but I can't locate anything in code and/or traces. Can you point me to what I submitted?

keesverruijt, 2025-05-08:
> [@keesverruijt](https://github.com/keesverruijt) I'm trying to find what PGN 129833 is/was doing, but I can't locate anything in code and/or traces. Can you point me to what I submitted?

Sure, git blame for PGN 12833 points to pull request  #360 .

keesverruijt, 2025-07-20:
See #572

closed 2025-07-20 by keesverruijt