canboat/canboat #530: Split GEOGRAPHICAL_COORDINATE into 2 types

Elmue, 2025-05-05:
I implemented the option for the user that he can click on geographical coordinates and the browser will open and show the location in Google Maps:

[image]

Here you see a Longitude and Latitude and a tooltip when the mouse is over one of the 2 fields.

The problem is that 
`<FieldType>NUMBER</FieldType>`
and
`<PhysicalQuantity>GEOGRAPHICAL_COORDINATE</PhysicalQuantity>`

do not allow me to distinguish between Longitude and Latitude.

Currently I have to parse the name of the field which is not very safe.

```
switch (Signal.DispName)
{
    case "Latitude":
    case "WP Latitude":                 // WayPoint (may be cloned multiple times!)
    case "Destination Latitude":        // Navigation Data
    case "Latitude of Vessel Reported": // DSC Distress
    case "Station Latitude":            // Tide Station Data
        // Store latitude ....
        break;

    case "Longitude":
    case "WP Longitude":                 // WayPoint (may be cloned multiple times!)
    case "Destination Longitude":        // Navigation Data
    case "Longitude of Vessel Reported": // DSC Distress
    case "Station Longitude":            // Tide Station Data
        // Store longitude ....
        break;
}
```

It would be useful if GEOGRAPHICAL_COORDINATE
would be replaced with
GEOGRAPHICAL_LONGITUDE
and
GEOGRAPHICAL_LATITUDE
so I can remove the `switch()` which depends on the parameter name.
__________________________________

In PGN 129027 there are fields "Latitude Delta" and "Longitude Delta" which are also marked as GEOGRAPHICAL_COORDINATE.
But these are not useful for opening them in Google Maps.

So they could have a PhysicalQuantity like GEOGRAPHICAL_OFFSET or something like that.

keesverruijt, 2025-05-08:
I guess you could look at the ranges, but this is not such a bad idea and we're doing a major version bump anyway so I guess this is OK.

Elmue, 2025-05-09:
> I guess you could look at the ranges,

No it cannot:

```
<Field>
  <Order>4</Order>
  <Id>latitude</Id>
  <Name>Latitude</Name>
  <BitLength>64</BitLength>
  <BitOffset>56</BitOffset>
  <BitStart>0</BitStart>
  <Unit>deg</Unit>
  <Resolution>1e-16</Resolution>
  <Signed>true</Signed>
  <RangeMin>-922.3372036854776</RangeMin>
  <RangeMax>922.3372036854776</RangeMax>
  <FieldType>NUMBER</FieldType>
  <PhysicalQuantity>GEOGRAPHICAL_COORDINATE</PhysicalQuantity>
</Field>
<Field>
  <Order>5</Order>
  <Id>longitude</Id>
  <Name>Longitude</Name>
  <BitLength>64</BitLength>
  <BitOffset>120</BitOffset>
  <BitStart>0</BitStart>
  <Unit>deg</Unit>
  <Resolution>1e-16</Resolution>
  <Signed>true</Signed>
  <RangeMin>-922.3372036854776</RangeMin>
  <RangeMax>922.3372036854776</RangeMax>
  <FieldType>NUMBER</FieldType>
  <PhysicalQuantity>GEOGRAPHICAL_COORDINATE</PhysicalQuantity>
</Field>

```

closed 2025-05-18 by keesverruijt