canboat/canboat #531: PGN 130845 - Simnet: Key Value

Elmue, 2025-05-06:
1.)
I study this PGN and I don't understand why there is a parameter named "MinLength" ?

[image]

This is from logfile susteranna-actisense-serial.raw
I see that there is a communication between the controllers 0C and 0E.
Controller 0C sends a request to 0E, which is marked by a length of 0 bytes.
A few milliseconds later controller 0E sends a response with the requested value: 3009  (0x0BC1)
The response has exactly 2 bytes as indicated by the parameter "MinLength"

So my question is why is the name "MinLength" instead of "Length" ?
In the other PGNs 130824, 130833, 130846, which also contain Key-Value pairs, the length parameter is called simply "Length".
Why is it different here ?
Where is the "Minimum" here ?
I see that the exact Length count of bytes is transmitted.

2.)
I recommend to rename "Address" into "Destination Address" which makes this parameter clearer.

keesverruijt, 2025-05-08:
It's called MinLength because I also see messages where the data seems to be four bytes with the length field saying two.

I would categorise this PGN as "not fully understood yet." Until someone makes the effort to dig deeper I don't think we should change anything yet.

keesverruijt, 2025-05-08:
I checked again for #529 and the MinLength field is really a mystery right now, for this PGN the lengths are all in the lookup table it seems. Why is that minlength field there? No idea right now.

See #529 changes, and try my analyzer on `samples/simrad-ap48.raw` and notice that some lookup values are not correct yet.

Elmue, 2025-05-09:
Is it  possible that you put the name "MinLength" because you saw messages with a Length field of 0 bytes ?
I see in multiple logfiles that 0 bytes means that it is a request to another controller.

Elmue, 2025-05-09:
> It's called MinLength because I also see messages where the data seems to be four bytes with the length field saying two.

For a reason I selected the screenshot that you see above.
The response message shows exactly this situation:
```
02 : Length byte
C1 0B : Two data bytes
00 00 : Two additional bytes
```

How is this to interpret ?
If the four bytes C1 0B 00 00 would all be data bytes, then the value of the Length byte 02 would be wrong. 
It would have to be 04.

But if we think a bit deeper about it, something like a "Minimum Length" byte **cannot** even exist.
How does the receiver interpret the "minimum" length of the data ?
The receiver must always know the **exact** length to decode the data correctly.
The minimum length is useless for the receiver.
It is completely impossible that the interpretation of a message is left to the receiver.
The sender must always tell the receiver how to decode the packet.
And if the sender sends wrong data the receiver will decode a wrong value.

Therefore all variable-length strings have a length byte.
Wherever you have variable-length data a Length byte is **indispensable**.
A variable length string with a wrong length byte can never be decoded correctly.

The same applies to PGN 130845.
Imagine the Length Byte is 2 and then follow 4 data bytes: AA BB CC DD
If the receiver respects the Length byte it will decode the data to a value of 0xBBAA and ignore the left over bytes.
If the receiver ignores the  Length byte it will decode the data to a value of 0xDDCCBBAA which is a completely different value.
So what does the receiver do when there is a contradiction between the Length byte and the count of data bytes?

If the receiver is supposed to ignore the Length byte and to use all the four following data bytes, why is a Length byte sent at all? They could leave the Length byte away and send only the data.

There is only ONE conclusion possible:
Something like a MinLength byte CAN not even exist.
It would make no sense,

_______________________

There is only one possible explanation for the 2 additional bytes at the end:
A sloppy firmware developer sends useless bytes to CAN bus.
I have seen SO many messages in your logfiles that show sloppy programming and breaking the rules of CAN bus.

The best example is logfile pgn129808.raw where they send a String_LAU like this:
12,01,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF
They send 18 bytes to transmit an empty string. 
To send this empty string 3 CAN bus packets are required!
What kind of software developers did this company hire?
They surely don't make IQ tests when they chose their employees.

I have never seen so many useless bytes sent in the J1939 world as I see in the NMEA 2000 world.

I have reverse engineered the assembler code of ECUs and dived very deep into firmware programming.
I find functions that are never called.
I find values being calculated and stored in a variable that are later not used anywhere.
I find security holes in algorithms for ECU flashing that require a key to unlock the programming mode. But you can send a wrong key, the ECU responds with an error code, however the ECU is unlocked for programming although the key was wrong.
I see so many projects that people upload to Github that are full of bugs and ugly code.
We live in a world of software that is full of bugs.

So I'm not surprised that we find logfiles with 2 useless filler bytes at the end of a message.

Elmue, 2025-05-09:
I made further investigations and found that the byte "MinLength" is completely useless.
It is either a Length byte with a wrong value (always 1) or it has another meaning.

Here my result with logfile simrad-ap48.raw where I compare the "Bits" attribute in the XML file with the Length byte:

[image]

B&G sends a valid length byte.
Simnet always sends 1.

A solution is to ignore the Length byte from the message and use the length in the XML file.
But this is also wrong in some cases like in PGN 130846:

[image]

Here 8 bytes are sent but XML says:
`<EnumFieldType Value='32789' Name='Alert bits' FieldType='BITLOOKUP' Bits='16'>`

It is required to print the contradictions between XML and real messages in all logfiles and adapt the "Bits" in the XML to the correct values, because the (Min)Length byte from Simnet is garbage.

Elmue, 2025-05-09:
Finally:
I found out that the field "MinLength" is definitely a Length field that contains valid data in logfile simrad-p9.raw

```
Mesg PGN 130845 'Simnet: Key Value' , Key 0:    field MinLength= 16 bit is same as XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 768:  field MinLength= 16 bit is same as XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 1024: field MinLength= 16 bit is same as XML LookupFieldTypeEnum= 16 bit

```
It is always 1 in logfile simrad-ap48.raw

```
Mesg PGN 130845 'Simnet: Key Value' , Key 41: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 520: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 32 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 264: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 32 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 1800: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 517: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 260: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130845 'Simnet: Key Value' , Key 516: field MinLength= 8 bit contradicts XML LookupFieldTypeEnum= 16 bit

```
So my conclusion is that the device ap48 has a buggy firmware where they always send the length of 1 byte.
Device p9 has a correctly working firmware where the Length matches.

So the field "MinLength" can be renamed to "Length" with a comment explaining that some buggy firmware may send wrong values in this field.

____________________

On the way I found that B & G is mainly correct.
But in logfile susteranna-actisense-serial I also find contradictions:

```
Mesg PGN 130824 'B&G: key-value data' , Key 60:  field Length= 32 bit contradicts XML LookupFieldTypeEnum= 16 bit
Mesg PGN 130824 'B&G: key-value data' , Key 158: field Length= 32 bit contradicts XML LookupFieldTypeEnum= 16 bit

```