canboat/canboat #540: Are we sure that 3 bits number has "unknown" value?

tomer-w, 2025-05-25:
In my system I get PGN 60928 - ISO Address Claim which looks like this:
```2025-04-10T12:10:16.614Z,6,60928,5,255,8,f5,01,c0,2c,ef,aa,46,c0```
The output of this is:
```
2025-04-10T12:10:16.614Z 6   5 255  60928 ISO Address Claim:  Unique Number = 501; Manufacturer Code = Victron Energy; Device Instance Lower = Unknown; Device Instance Upper = 29; Device Function = Battery; Device Class = Electrical Generation; System Instance = 0; Industry Group = Marine; Arbitrary address capable = Yes
```
It all makes sense except the ```Device Instance Lower = Unknown``` part. Are we sure they meant it to be unknow? it makes more sense to be 7. I think that at least some 3 bits number do use all the bits.

What do the NMEA 2000 experts think about this example?

aldas, 2025-05-25:
I tested that payload with my own Go code and `Device Instance Lower` parses to `7`. and that code is based on Canboat (v4.10.0).

keesverruijt, 2025-05-27:
Would

    <ErrorValue>254</ErrorValue>
    <UnknownValue>255</UnknownValue>

be better? I think so.

tomer-w, 2025-05-27:
yes, I think this is way better option.