canboat/canboat #541: Check Primary Key fields

keesverruijt, 2025-05-26:
PGN 60928 ISO Address Claim has primary key fields that are not appropriate: it is not true that a device with a particular source address will send out multiple claims with different values for these fields.

Check all Primary Key assignments.

tomer-w, 2025-05-26:
hi @keesverruijt , I will be glad to fix something if needed. Saying that, I'm not sure I understand the issue here.

With the introduction of primary key, the PGN fields are split to 2 groups:
1. primary keys
2. readings / sensors values.
The way to use this field in the general case (not this PGN) is to generate a gauge / sensor based on the hash of the relevant ISO Address Claim for this address (I keep a dictionary of source to ISO Address Claim) and all primary keys (#1) combination and the sensor values based on the rest (#2).

This logic works well for all regular PGN. In this special PGN, all fields are part of the primary key as they will never change and the only "reading" is the source address which will change based on what address is available on the network. The only thing which is unique is that this source address is not part of the PGN definition. but I dont think it breaks the model.

This is all assuming I got the usage of this PGN right, if I have some wrong assumption here, I will be glad to learn more about the subject.

tomer-w, 2025-05-26:
Yes, if there is a takeover we want the mapping to change. This is exactly my implementation. But the thing which was changed - (the sensor value / reading) is the source ADDRESS. The NAME is the constant part and therefore it should be the primary key. The NAME is all primary key as it describe a single device in a constant way. Whatever the source address is. It is just unique as in this PGN there is nothing else as part of it.

Anyhow, if this is not convincing you, I will prepare a PR to remove the primary key from all fields in this PGN. I think it might require some work as this PGN is using the INSTANCE_FIELD macro which always assume primary key (another hint that it a is real primary key). My code anyhow handles this PGN in special case so it will not break anything.

TwoCanPlugIn, 2025-05-27:
An interesting discussion. 

If one were to think in purely database terms, I would have thought that the primary key is the source address. You can't have a duplicate address, and it uniquely identifies the source of a message. It is quite common for a single physical device, for example a B&G NAC-3 Autopilot Computer to support different NMEA 2000 devices (Autopilot, NMEA-0183 gateway, Rudder Sensor) each claiming their own address and each with a different NAME. 

On a similar vein, and although unlikely to occur, I don't believe there is anything that prevents the same device with the same NAME from claiming multiple addresses. 

Whilst addresses may change due to the "address claim dance" it doesn't change the fundamental rule that there are no duplicate addresses.

To use Kees' database analogy, think of a SQL database, one table for devices (derived from a combination of PGN's 60928 and 126996), and another table for data for PGN's that encode sensor readings.

Putting this into a series of SQL statements, one could envisage something like this:
```
CREATE TABLE IF NOT EXISTS devices (source INTEGER PRIMARY KEY, name INTEGER NOT NULL UNIQUE, modelid TEXT, productid TEXT);
CREATE TABLE IF NOT EXISTS data (source REFERENCES devices(source), pgn INTEGER NOT NULL, payload BLOB NOT NULL, receipt TEXT DEFAULT (STRFTIME('%Y-%m-%d %H:%M:%f', 'NOW')) );
```
On receipt of PGN 60928, the devices are inserted into the table:
```
INSERT INTO devices (source, name) VALUES (0,1234);
INSERT INTO devices (source, name) VALUES (2,4567);
```
Upon receipt of PGN 126996 each device is updated with product info.
```
UPDATE devices SET modelid = '508 Wind Sensor', productId = '29579' WHERE source = 0;
UPDATE devices SET modelid = 'DST810', productId = 'IDST810-NV40-OEM0' WHERE source = 2;
```
And then upon receipt of PGN's with sensor readings, they are added to the data table.
```
INSERT INTO data (source, pgn, data) VALUES (0, 130306, X'0b4f007ea2faffff');
```
Finally you could then issue a query to retrieve the source address, modelid, productid and data using a query similar to the following:
```
SELECT data.receipt, devices.source, devices.modelid,  data.pgn, data.payload FROM data  INNER JOIN devices ON devices.source = data.source WHERE data.pgn = 130306 ORDER BY receipt;
```
In my mind it seems quite obvious that the source address is the primary key.

If the device is reporting multiple instances, if you really want to specifiy a key, just make a composite key that spans both the source address and instance number. However the problem is that not all PGN's have an instance value.  PGN 127506 DC Status does, whereas PGN's 130306 Wind and 126287 Depth don't. I guess that reflects the nature of the equipment, most battery chargers support multiple battery banks.

tomer-w, 2025-05-27:
Hi, @TwoCanPlugIn , thanks for chiming in.

I think we first need to discuss the scenario behind this primary key requirement to understand what analogy fits best.
I want to have a specific gauge and accompanied history chart of readings for each sensor reading coming up from PGNs. It should be as stable as possible and for sure survive network restarts. For this I need a primary key which is as stable as possible by the design of the NMEA 2000 protocol.

The analogy I have is to the mapping between my bank accounts IDs and the amount of money I have in each. I assume that neither the bank nor my applications will use the IP address of my computer as a key to map to my details.
The source address which is part of the NMEA 2000 protocol should be considered at a different protocol layer than the NAME. the address is transient while the NAME is constant.
As the NMEA 2000 protocol use a slow network, they are not sending the NAME which is equivalent to my bank account ID or authentication cookie in each request but on intervals. This means that my application (or any other similar implementation) will ignore received packets before it gets the mapping between source address and NAME I as I have no idea where to store this new data. Only after I have the ISO Address Claim PGN received I know in what row of the table to put it in (going back to your SQL analogy).

So in my opinion the primary key for a sensor reading is the combination of:
1. PGN ID
2. ISO Address Claim NAME
3. Hash of all primary keys fields.
The source address needs to be stored in memory and not in the tables. I do need it just to be able to have `#2` available when I do want to store the value, but it is not part of the SQL storage.

For that reason, I also think that all the data which is constant (items```#1``` - ```#3```) should be marked as primary keys as they are not really sensor reading. Specifically, as ISO Address Claim PGN is special anyhow, I dont really mind how it is marked as it will not get into the long-term storage tables we are just talking about, just for consistency I did want them to be flagged this way.

keesverruijt, 2025-05-27:
> This means that my application (or any other similar implementation) will ignore received packets before it gets the mapping between source address and NAME I as I have no idea where to store this new data.

Which is why, on startup, you should do an ISO request for address claims so you receive the NAMEs as soon as possible. See `util/list-product-information` for a PHP script that shows an example of this.

keesverruijt, 2025-05-27:
Also, using the NAME as the real identity of a device is "the proper way." In practice though my source addresses on my boat's bus are pretty stable and I swap out devices quicker than the address claims change ;-)

tomer-w, 2025-05-27:
> Which is why, on startup, you should do an ISO request for address claims so you receive the NAMEs as soon as possible. See `util/list-product-information` for a PHP script that shows an example of [...]

I fully agree. I'm currently not doing that as I'm just getting started. It is also not super important for me if after an application restart, I will lose a few more seconds to build the database but for sure I need to do that if I want it to be right.

tomer-w, 2025-05-27:
yes, I will keep improving on getting the NAME re-sent by using the PGNs you mentioned. With regards to the need for defining primary keys to begin with, I am super happy Kees accepted this concept as I do see tons of value of storing the meaning and any information we have on PGNs centrally, for the sake on the community and not having islands of knowledge spread between different repos in different format. That is what makes canboat project so unique to begin with.