canboat/canboat #545: Quality of Logfiles

Elmue, 2025-05-31:
Hello Kees,
I used all your logfiles for testing my code.
And I found that I need additional information for some files.

It is obvious that files like 129540.raw, 130311.raw, pgn130330.raw have been edited manually.
But some of the others need explanation from you.

I implemented a parser into HUD ECU Hacker for all logfile formats that are here on Github.

__________________________________

First question:
The files in EBL format follow one of the two schemes:
```
DLE STX <command> <len> [<data> ...] <checksum> DLE ETX
or
ESC SOH <frametype> [<data> ...] ESC LF
```

The only file that does not follow this scheme is pgn1.ebl which has this content:
```
03 20 10 E7 A7 84 83 D9 01 
1B 01 - 07 95 0E 28 9A 20 01 F8 09 3D 0D B3 22 48 32 59 0D - 1B 0A  
```

The second line complies with the second scheme.
But the first line does not use none of the schemes.

So my question is:
Is this a manually created file?
Is this file correct or incomplete?
Are there 4 bytes missing in the first line?

Is it allowed that a sequence of bytes appears without enclosing 
```
1B 01 ..... 1B 0A  
or  
10 02 ..... 10 03?
```

I suppose that the file is corrupt because EBL Reader shows the time in red:

[image]

_______________________________

Logfiles are a very important source for testing.
They are even indispensable.

It would be really helpful if you would upload a file Readme.txt that explains for each logfile:

1.) The hardware that has recorded this file,
2.) The software that has recorded this file,
3.) A list of all the devices (manufacturer + model) that were connected to CAN bus,
4.) If the file has been manipulated manually.
5.) Some more info about the environment, for example the real water depth to prove the correct display of the values.

There is a logfile "garmin-fake-126720.raw"
Why this name?
What is fake in this file?
An explanation would be useful.
______________________________

There are several logfiles with corrupt time stamps.
The worst is dirona-actisense-serial.raw
It has 723 errors.
When I load it with my logfile parser I get:

```
The logfile contains inconsistent timestamps. The time is jumping backwards in line 171.
[... 295 lines cut ...]
Not even the important Request / Command / Acknowledge Group Function is implemented.

[image]

Actisense is not able to display the source address correctly.

Also the Acknowledge Group Function is decoded totally wrong.

The next screenshot shows a field "Fields 4 - 12 repeat as needed".
And where are they ?

[image]

And this happens when a message contains Unicode strings:

[image]

When I load a DSC Call, Actisense shows me binary data.

Why is Actisense selling an expensive NGT-1 adapter and betraying the clients with such a crap software?
The answer is simple:
Nobody will ever notice this because nobody has the NMEA documents and can prove if they show correct or wrong values.
The fact that the NMEA keeps all information secret allows them to sell sloppy software and their clients will even be happy because they have no chance to ever find out.
I'am the first one who did such a deep analysis.
May light come into the darkness!

sbender9, 2025-06-01:
I am glad you read all of that, not sure I could 🤣

sbender9, 2025-06-01:
Ok. I should speak up too.

@Elmue  you are spending way too much time complaining and bitching and moaning. 

My advice to you, would be to fork the project and make it better. 

Get back to us when you address all of this.

woobagooba, 2025-07-19:
@keesverruijt what are you waiting for?