canboat/canboat #550: FUSION_REPEAT_STATUS Lookup Enumeration has incorrect MaxValue of 0

meaning-matters, 2025-07-02:
In canboat.json version 6.0.1:

```
      {
        "Name":"FUSION_REPEAT_STATUS",
        **"MaxValue":0,**
        "EnumValues":[
          {"Name":"Off", "Value":0},
          {"Name":"One/track", "Value":1},
          {"Name":"All/album", "Value":2}
        ]
      },
```

keesverruijt, 2025-07-07:
Hmm, these are four byte lookups, I haven't looked further but my guess is that this is overflowing a 32 bit integer somewhere, certainly on 32 bit systems. The expected MaxValue is 2^32 - 3. 

I'm not sure where you get "maximum integer field value, but this doesn't apply for BITLOOKUP" from. (a) the maximum integer value is true for bitfields as well just without the "no data" / "unknown" values, (b) these aren't bitfields.

meaning-matters, 2025-07-08:
About BITLOOKUP, here's the first one:

```
      {
        "Name":"STATION_STATUS",
        "MaxValue":3,
        "EnumBitValues":[
          {"Name":"Station in use", "Bit":0},
          {"Name":"Low SNR", "Bit":1},
          {"Name":"Cycle Error", "Bit":2},
          {"Name":"Blink", "Bit":3}
        ]
      },
```

With 4 bits, the maximum integer value would be 2^4 - 1, not 3. This is what made me wonder what `MaxValue` meant (and what its purpose is).

commit 0bc860ec (2025-07-12): Fix #550
closed 2025-07-12 by keesverruijt