canboat/canboat #557: Errors in Canboat.xml

Elmue, 2025-07-11:
I automatically verified the Bit Length sent in all messages in all logfiles 
and compared them with the Enum Bits="XY" in XML.

"B&G Key Value" sends valid DYNAMIC_FIELD_LENGTH in all messages in all logfiles.
Bits="XY" is not required in XML for BANDG_KEY_VALUE.

But for SIMNET_KEY_VALUE it is important that the Bits length in XML is correct 
because Simnet does not send the length of the data in the message.
Byte 9 (Order 10) in PGN 130845 is neither Length nor MinLength. 
It is ALWAYS 1.
This field can be changed into RESERVED. It has no useful information.

Check with logfile simrad-ap48

```
0DFF1D07: 41 9F FF FF 02 FF 29 00 00 01 - 88 - FF FF FF
<EnumFieldType Value="41" Name="Timezone offset" FieldType="DURATION" Signed="true" Resolution="60" Unit="s" Bits="16" />
```
Bits="16" should be Bits="8". The last 3 bytes are FF
The value 0x88 = -120 minutes makes totally sense, while 16 bit would be 0xFF88 = -78 minutes time offset which does not exist.

```
0DFF1D07: 41 9F FF FF 01 FF 08 07 00 01 - 00 - FF FF FF
<EnumFieldType Value="1800" Name="Anchor depth" FieldType="NUMBER" Signed="false" Resolution="0.01" Unit="m" Bits="16" />
```
The last 3 bytes are unused FF as in many other messages in the same logfile.
Is this a casuality that the ancher depth is just exactly 0xFF00 ?
This would mean 65280 * 0.01 = 652.8 meter.
Or is this 8 bit and Resolution is 1 meter which means anchor depth = 0 meter?

```
0DFF1E07: 41 9F FF FF 01 FF FF 15 80 01 08 - 14 00 00 00 00 04 50 01
<EnumFieldType Value="32789" Name="Alert bits" FieldType="BITLOOKUP" Bits="16">
```
Bits="16" should be Bits="64".
SIMNET_ALERT_BITFIELD defines 57 bits, so Bits="16" is definitley wrong here.

__________________________

Additionally for
```
<PGN>65285</PGN>
<Id>lowranceTemperature</Id>
in
<Order>4</Order>
<Name>Temperature Source</Name>
the
<PartOfPrimaryKey>true</PartOfPrimaryKey>
```
is missing.

The Primary Key is also missing in all fields that are DYNAMIC_FIELD_KEY.

keesverruijt, 2025-07-19:
The PK issues are addressed in #541 (work in progress). Next time, don't think of issues as email: one problem per issue. That's why they're called issues.

keesverruijt, 2025-07-19:
> Check with logfile simrad-ap48
>
> 0DFF1D07: 41 9F FF FF 02 FF 29 00 00 01 - 88 - FF FF FF
> <EnumFieldType Value="41" Name="Timezone offset" FieldType="DURATION" Signed="true" Resolution="60" Unit="s" [...]

Sorry, you are wrong. Examining the first 18 lines of `simrad-ap48.raw`, which I painstakingly made by pressing buttons on one of my AP-48s and then examing the PGNs it generates, shows that timezone offset is in minutes, and that it can go beyond 255 minutes. It can be set in multiples of 15 minutes. Ergo, it requires 16 bits. The logfile shows this clearly:

    # Set timezone to GMT+00:00, dd/mm/yyyy
    2023-06-09T18:12:51.545Z,3,130845,7,255,14,41,9f,ff,ff,02,ff,29,00,00,01,00,00,ff,ff
    # Set timezone to GMT+01:00, dd/mm/yyyy
    2023-06-09T18:13:54.344Z,3,130845,7,255,14,41,9f,ff,ff,02,ff,29,00,00,01,3c,00,ff,ff
    # Set timezone to GMT+13:00, dd/mm/yyyy
    2023-06-09T18:14:31.867Z,3,130845,7,255,14,41,9f,ff,ff,02,ff,29,00,00,01,0c,03,ff,ff
    # Set timezone to GMT-00:15, dd/mm/yyyy
    2023-06-09T18:15:29.919Z,3,130845,7,255,14,41,9f,ff,ff,02,ff,29,00,00,01,f1,ff,ff,ff
    # Set timezone to GMT-02:00, dd/mm/yyyy
    2023-06-09T18:15:38.552Z,3,130845,7,255,14,41,9f,ff,ff,02,ff,29,00,00,01,88,ff,ff,ff

Notice the 3rd occurrence, where we set 13 hours = 780 (0x30c). Notice how 16 bits are used.

The confusion is maybe because you forgot to convert hex to decimal? 0xff88 as a 16 bit signed value is -120, not -78 as you claim. Decimal -120 is hex -78 ...

For anchor depth and the alarm bitfield I need to do further reverse engineering to get to the bottom of it, and for better or worse I try to make changes to a minimum nowadays so it will have to wait until I get a chance to do so (the bitfield data was contributed, and for anchor depth I already note in the logfile that I did not see expected values.)