canboat/canboat #567: Projects that use Canboat

Elmue, 2025-07-15:
**Hello Kees**

You made a big work creating this database.
But did you ask the question for whom this work is useful?

Let's imagine an ordinary boat owner who wants to see what is going on the CAN bus of the boat.
What options does he have?

1.)
He can buy an expensive PC interface for example from Actisense for $250 that comes with a software like NMEA Reader that shows wrong information, does not implement many of the messages and has a user interface that shows only one message at a time.
He will not even see one of the messages that you have reverse engineered like form B&G, Simnet, Raymarine, Furuno, Airmar, Lowrance, Yacht Devices, etc.

2.)
He cannot and will not compile your analyzer C application if he is not a software developer.

3.)
Even if he compiles your C application he is probably not willing to learn all the clumsy command line parameters and see only text output in a console. We are living in 2025 and I don't know anybody who still uses command line tools.

4.)
So an ordinary boat owner who is not a software developer will not make use of your database. 
Only very few freaks who are boat owners AND also software developers can make use of it.

5.)
How can more people profit from your work?
My proposal is that you could add another page like for example
https://canboat.github.io/canboat/software.html
where you present a list of all the projects that use your database.

There could be a short description with some **screenshots** of each project that you know.
There could be a **download link** for the already compiled software, so also users who are not software developers can install it easily. 
It would be interesting to see an overview of all projects that use your database.

One of the projects on this page could be **HUD ECU Hacker** which is released now.
https://netcult.ch/elmue/hud%20ecu%20hacker
It has an installer which installs the software and your database with one click.

With a **cheap Tactrix adapter** for $20 US on Alibaba people can connect to the CAN bus of their boat.
(By the way: with the same adapter HUD ECU Hacker can also scan their car or motorbike)
Also the installer for the driver of the Tactrix adapter is included in HUD ECU Hacker.

The user can record logfiles which will have correct **timestamps** in contrast to many of the logfiles that I downloaded here.
The logfile format is optimized to use as less space on disk as possible.

The user can see the live data on the CAN bus in real time in a very comfortable **user interface**.
With HUD ECU Hacker they have high quality software that has been written by a software developer with 40+ years of programming and reverse engineering experience.

Apart from NMEA 2000 also 8500 messages from the **J1939 protocol** are implemented.

The **logfile parser** can read several logfile formats: 
Linux CAN dump, Wireshark, CSV, Actisense (ASCII + EBL), Yacht Devices, Garmin, iKonvert, PCAN View, ELM327 and HUD ECU Hacker's XML format. Logfiles are recorded and can be copied into subfolder "ECU\J1939\Logfiles". The button "Play Logfile" plays a logfile.

Experts can control HUD ECU Hacker by **marco scripts** that will be compiled before execution and run as fast as native C code.
The installation already comes with a macro script "Send Logfile" that can send any logfile to the CAN bus, which allows to emulate a device.
Also single CAN commands can be sent manually to the CAN bus in the CAN Raw Terminal.
Macro scripts can receive CAN bus messages, process them and send responses to the CAN bus. The possibilities are infinite.

And last but not least:
HUD ECU Hacker is **charityware**, which means that the author does not earn any money with it. 
He only asks for a voluntary donation to a non-profit organization of the user's choice.

[image]

[image]

meaning-matters, 2025-07-15:
I'm working on iPhone & iPad app based on CANboat that does advanced NMEA 2000 traffic display.

Yes it will require some NMEA 2000 adaptor/gateway like the Actisense W2K-1, so it might indeed be more for professionals. However, it will give visual access to the immense amount of data CANboat unlocks.

TwoCanPlugIn, 2025-07-18:
Your HUD ECU hacker tool looks like an excellent product and your experience and knowledge in developing software for motor vehicles and ECU's is clearly evident. For the marine industry and for decoding NMEA 2000 data, you are indebted to Kees and the Canboat project.  

I do hope you have realistic expectations, in that 99.99% of sailors are not interested in tools such as these, or for that matter Canboat. Probably little different to the 99.99% of car drivers who have no interest in their car's electronics.

However I do get a giggle from your previous comments in #533 

> But I have implemented several CAN adapters into my software which also use COM ports.
It is usual that these adapters transfer data in ASCII format over the COM port.
That means a data byte 0x8A on CAN bus is transfer as two ASCII bytes "8" and "A" over the cable.
Additionally they must transmit the PGN, Priority, Source, Destination or the CAN Bus address and they must transmit where a new packet starts and ends or the length.
All these bytes summed up result in a throughput far less than 230 kBaud.
If they use ASCII transfer they will perhaps get a throughput of 100 kBaud.
On a CAN bus with high traffic you will lose messages.

I've been using the Canable Cantact adapter for a long, long time. in fact whilst I can't take any credit for it, I did make the suggestion to Ethan to add galvanic isolation and to provide a M12 connector and shortly afterwards they released the Pro version, with isolation.

Your HUD ECU Hacker only connects to the Canable Cantact when it is a serial device using the original SLCAN firmware. For better performance why don't you support it as a native USB device using the Candlelight firmware?

For example using some publicly available libraries your interface requires only a few lines of code. 

`err = candle_init_single_device(deviceNumber, channelNumber, 250000, CANDLE_MODE_NORMAL, &list, &deviceHandle);`
`if (candle_frame_read(deviceHandle, &frame, 1)) {`
`	// Only interested in extended frames`
`	if (candle_frame_is_extended_id(&frame)) {`
`		DecodeCanHeader(buf, &header);`
`		AssembleFastMessage(header, frame.data);`

No serial port programming, no nibbling through strings to find delimiters, no parsing of ASCII strings such as: 

`"T1D11220381122334455667788\r";`

And better performance...

keesverruijt, 2025-07-19:
You can add your projects to the Wiki main page: https://github.com/canboat/canboat/wiki

keesverruijt, 2025-07-19:
And please, make sure you divulge your use of my (our) database in your software download page and in the software; especially if it is not open source. 

Why isn't your software open source?

Elmue, 2025-07-19:
**Hello TwoCanPlugIn**

> I do hope you have realistic expectations, in that 99.99% of sailors are not interested in tools such as these

Yes I know.
By the way I do not have any "expectations" at all, because I do not earn money with my software. It is my hobby. 
It is not up to me what people do later with my software.

As I already implemented all the framework for J1939 before coming to Canboat, it was not such a big step to add some more messages for NMEA. I just completed the work so all officially known messages can be decoded.

I also suppose that more people in repair workshops of trucks, buses or agricultural vehicles will use the J1939 messages than sailors will use the NMEA messages.

__________________________________

> why don't you support it as a native USB device using the Candlelight firmware?

This may come in a future version.
All what I have functioning so far was already an eternal work.
Alone J1939 and NMEA took me half a year to implement.
Over 10 thousand lines of complicated code.
And my users ask me "Could you add this ECU model from Yamaha or this ECU from Harley Davidson?"

So HUD ECU Hacker is a project that occupies all my time and will never be finished.
I have to set priorities what I do next.

I would also have to add a firmware updater that allows my users to update their CANable with one click. 
I don't like the current way of firmware update over a website which is working only in the Chrome browser.
https://canable.io/updater/canable2.html

What if this website is not available anymore tomorrow ?
I already have the installers for all adapter drivers implemented, I would have to add this firmware updater too, in some future. Currently this does not have priority, but some day it may come.

________________________________

**Hello Kees**

> https://github.com/canboat/canboat/wiki

OK. I will check out this Wiki.

> And please, make sure you divulge your use of my (our) database in your software download page and in the software; 

OK. No problem.
However your original comments 

```
<Copyright>CANboat version v6.0.1

(C) 2009-2025, Kees Verruijt, Harlingen, The Netherlands.
For more information see https://github.com/canboat/canboat

This file is part of CANboat.
```
are maintained so everybody who is interested where the XML file comes from (that I put into a ZIP file) will easily find this information.

______________________________

> Why isn't your software open source?

One reason is that I made bad experiences in the past with open source projects.
It happens that developers have different opinions of how to implement a functionality.

Then they make a clone on Github and suddenly my software exists ten times with little variations.
And the user who wants to install HUD ECU Hacker does not understand what is the difference between all the forks and which to download because most people on Github are so lazy that they don't even write a description of what their project is about. And they will introduce new bugs into my perfectly tested code and so on and so on.....
There would be one version that I continue to develop and multiple other versions that are old and buggy.
I can only say: No, thank you!

Some examples for different developer opinions have also happened here when I made proposals that you have rejected.
For example you use many abbreviations in your database that are difficult to understand like "Used+Diff" in SATELLITE_STATUS. What does "Diff" mean?
You may know what this means, but will my users also understand this?
So I made my own database that displays "Used with differential corrections".
This is the official name in the NMEA 2000 Appendix B that you have abbreviated for any reason.

Originally I had the idea to let all the NMEA messages be based only on your XML file.
But when you started to make weird changes like converting kWh into Joule I noticed that it will not be possible to use a database where my users can download a new XML file and easily replace it because tomorrow it will be completely different.
So I ended up to create my own embedded database which contains all the official NMEA messages with longer text descriptions and using the external Canboat XML file only for all the non-official vendor-specific messages.
The official J1939 messages like ISO Address Claim where already implemented in the embedded database before I started with NMEA.

If an agreement about such simple things is already difficult, what would happen if my software was open source and others would be working on my code? I write high quality software, but many others write quick and dirty code. This is my experience from participating in open source projects in the past and it resulted in endless discussions which are boring and consuming my time, mostly fruitless.

But the **main reason** for not making it open source is that ECU flashing is a very lucrative business.
Many companies around the world sell expensive tools for ECU flashing.
For example the italian company Alientech sells the tuning tool KESSv2 for $9000.
Much money is moved in the tuning market.
If I would publish the source code of HUD ECU Hacker (which is charityware and by purpose affordable for every hobby tuner) these companies would steal my code, put it into their software and sell it for an extremely exaggerated price to their clients. I did not work all these years on a charity project to allow this.

And HUD ECU Hacker has features that NO other software in the world offers.
[... cut ...] this requires a very deep analysis on the ECUs assembler code. This allows me to extract the 800 calibrations with full information for ANY firmware version. This is software development at the highest level, resulting in features that only HUD ECU Hacker has. It took me one year to implement this. If HUD ECU Hacker would be open source several companies would have already stolen this code, because they will not invest so much work and time like I did. They do it the quick and easy way.

[image]

closed 2025-07-20 by keesverruijt